makerbase-mks/MKS-TFT #3: Filament Change

DR-DS, 2016-12-06:
Could you implement the filament change gcode for multicolor prints?

DR-DS, 2016-12-07:
It should : 
-Read the GCODE M600.
- Lift the extruder and move it to a corner.
- Retract the filament X mm (configurable in the config file).
- Show a screen with two buttons: "Extrude" and "Resume Print", so you can extrude until the new color is extruded cleand and you can resume the print.

It should be very easy to add and it's very useful.

Thank you very much.

makerbase-mks, 2017-02-07:
we will add this function in V2.0  before 17.2.20.
thank you very much

Morepizza1, 2017-03-17:
My filament sensor works perfect with an 12864 LCD:
https://www.aliexpress.com/item/12864-LCD-Ramps-Smart-Parts-RAMPS-1-4-Controller-Control-Panel-LCD-12864-Display-Monitor-Motherboard/32693237483.html

But I have now done an upgrade to a MKS TFT28 display (first with firmware v1.2 and afterwards with the latest firmware v2.0).

Everything works great but when I test the runout sensor, it doesn't work. The extruder moves to a predefined point and starts extruding. After this I can´t let the extruder take filament in again. There is no filament screen popping up like with the 12864 lcd.

When I put the screen in pause, I can go to the filament change screen, but I cannot feed filament in.

First I was thinking, it is the Marlin firmware, but with the 12864 lcd it works perfect.

Morepizza1, 2017-03-17:
Extra info:
My configuration:
MKS GEN v1.4 + MKS TFT28 (latest firmware v2.0: MKS-TFT_V2.0.0English/red style/classic)
I am testing with Marlin 1.1.0 RC7

veedubb65, 2017-03-29:
I've installed the v2.0 firmware (classic red) yesterday afternoon and it still doesn't respond to M600 when printing from the SD card. This would really be a huge help if it could be implemented. This was with the MKS TFT28 and MKS Base running Marlin.

makerbase-mks, 2017-06-27:
"Filament Change" is not good enough in V2.0,we felt sorry for that.
We will release V3.0 firmware before 7.22. In this version, we will optimize  "Filament Change" function.
The funtion is:
1. Click "Filament Change", printer pause and move to corner(position can be set in mks_config.txt)
2. Show a screen with four buttons: "In(Extrude)" and "Out(Retract)","E0"(select Extrude), "Back(Resume print)"
if Extrude temperture is below limit_temperature(set in the mks_config.txt)), auto heat extrude.
--->Click "In" can Extrude Filament  X mm (set in the mks_config.txt)).
--->Click "Out" can Retract Filament  Y mm (set in the mks_config.txt)).
--->Click "E0" can Select Extrude if have two Extrude.
--->Click "Back"  resume print.

GorillaPartsCo, 2017-07-22:
Is V3 out somewhere? It's 7/22, and I'm hoping it's been released. My main printer is down. I have an important project that needs to be finished, and the only machine I have that's functioning is running the MKS TFT32, and the pause feature doesn't work.

Morepizza1, 2017-07-23:
I have yesterday written a request to shelly@makerbase.com.cn and he replied with a v2 version included.
You can try it. Another option is to use an 12864 LCD. This is plug and play compatible with an MKS board, it works on my board immediately after plug in. On that lcd the pause function works correct.

Morepizza1, 2017-07-26:
Shelly has solved the problem and will post the solution. In short:
- for the pause/filament change function you have to use the connection of pin B1 on the tft board (see manual of tft) instead of the connection on the main board. 
- In the mks_config.txt firmware file you have to change the X and Y pause position from value -1 to value 10. 
Shelly will post a new firmware and more explanation.

GorillaPartsCo, 2017-07-27:
I look forward to the update.

closed 2017-08-02 by makerbase-mks

GorillaPartsCo, 2017-08-17:
Why was this closed?  This didn't solve my issue. My only solution is to run Octoprint on a Raspberry Pi.

AndreVanKammen, 2017-09-17:
I've just installed V3.0 and the M600 gcode is still completely ignored. Is there any other option to put the MKS display in pause mode from gcode?

I tried M0, but that put's the MKS_Base (Marlin) board in pause mode and can't resume it. This is probably because that function is ment to be resumed with the 20x4 display controller which i didn't get with my printer because i choose the MKS TFT. Maybe i can find the right pin to simulate the click?

It would be better if the MKS display just filtered out the M0 command and put itself in pause mode.

simonvez, 2018-03-14:
I would like to that M600 triggered when PB0 is triggered from the filament runout. right now it just pause, lift nozzle..and resume when I want.. but I would like to have more control over that. My M600 in marlin is well defined for my need. I need this :)

Nitrosito, 2018-08-17:
works or not?

Wallaby07, 2019-07-21:
Has this been fixed, maybe the SD card direct to the MKS-TFT will ever work if it gets fixed in the firmware.
I want to use this plugin [https://www.thingiverse.com/thing:2077884](url) it can work with the M25, M0, M600 commands?

marce025, 2019-08-05:
Version 3 of tft32 still does not work with the M600 command ... how can it be fixed?

kaliboe, 2019-11-08:
Hi all, I have the same issue, DR-DS started this topic almost 3 years !, is there a working solution for the M600 code?