makerbase-mks/MKS-TFT #4: Status Screen

EberleG, 2016-12-07:
Can you add a status screen that shows the current Z, X and Y, Hotend temp, Bed temp?  Or open up the source code so someone else can?

Steamerzone, 2016-12-11:
A status screen is also one thing that I'm missing, so much so that I'm considering going back to a 12864 LCD screen. Please consider adding it, or let someone else do it by opening up the source.

pmbroth, 2017-01-08:
Please give a status scree, it is needed!!!

AlexDoran, 2017-01-09:
I added a separate issue for this and another item, but this needs to be done or have the source code released, i have also sent them a message on Aliexpress.

Alex

aimartin, 2017-01-09:
It would be really awesome if they release the code, but honestly, I don't think they will do it :(.

+1 in this!

AlexDoran, 2017-01-09:
Guys, after some searching i have stumbled onto this guy's project who has been rebuilding the firmware himself, its seriously worth checking out:

https://hackaday.io/project/18852-alternative-firmware-for-mks-tft32

pmbroth, 2017-01-09:
He is just changing the button images, and added a few M commands for the leveling buttons.  I have this.  nothing else is going on, unfortunately  at this time.   I really want the status screen, such that I ordered an 12864 screen to have that as my status display, while using this to control the printer

aimartin, 2017-01-09:
I saw that project, and I've subscribed to it, although I don't think is mature enough... if you plan to use the screen as source for controlling the printer (executing gcode), not sure if that is trustable or not... However, is really interesting indeed, let see :)

AlexDoran, 2017-01-09:
@pmbroth I think he is rebuilding the firmware using his own source code, the images can be replaced using the standard firmware.

@aimartin Yes it appears to be in the early stages, maybe if there is enough interest diverted to the page it could be an incentive for him to really get it going. I wish i had the knowledge myself to undertake such a task.

pmbroth, 2017-01-09:
ok cool, then I misspoke.  Apologies!

makerbase-mks, 2017-02-07:
we will add this function  in V2.0 before 17.2.20. thanks

AlexDoran, 2017-02-07:
Excellent news, thanks for taking our feedback on board. Can't wait to try it out.

Alex

pmbroth, 2017-02-09:
Excellent News!!!  Can you look into the pausing issue that many are having when printing from sd card when having a filament sensor connected.

makerbase-mks, 2017-02-13:
maybe the filament sensor send wrong signal.

pmbroth, 2017-02-13:
If you look at the open issue for pausing, it seems it is not related to one printer, but several that are trying to use this feature.

snoerenberg, 2017-02-19:
Hi, when will the firmware 2.0 be released? On 17.02. I could not find any new FE here.

Thanks

AlexDoran, 2017-02-19:
King Zhong at MKS has sent me the new firmware, i can email it to you if you'd like.

Thanks

snoerenberg, 2017-02-19:
Can you upload it to zippyshare or something?
Thanks

makerbase-mks, 2017-02-20:
we have upload at: MKS-TFT/Firmware/MKS-TFT_V2.0.0English.zip; use  red or blue style 's retro firmware.
Add some new function,the datasheet is traslating, and will be upload after some day.

pmbroth, 2017-02-22:
The new screen is great!  Keep up the good work!

closed 2017-06-26 by makerbase-mks