makerbase-mks/MKS-TFT #7: Additional Feedback / Status Screen - Feature Requests

AlexDoran, 2017-01-09:
Hi there, 

I really like the TFT32 but i was wonder if it was possible to add some additional functions to its software. If i had a decent knowledge of coding i would attempt this myself.

First of all i think it would be nice for the Screen to Boot into a 'Status Screen', this would show the Current Temps / Fan Speed / Axis Positions (Much like the LCD RAMPS Screens do), from here you could navigate to the main menu.

Also, when a print is active it is really useful to be able to see current Axis positions, Especially the Z / Layer Height.

Are these features that could possibly be added?

Thanks

Alex

makerbase-mks, 2017-02-07:
we will add Temps/Z axis to status  in V2.0  .
thank you.

AlexDoran, 2017-02-07:
Perfect, Thank you.

Alex

nitr8, 2017-05-03:
Am i blind or is there no status in V2.0 @makerbase-mks

makerbase-mks, 2017-06-26:
MKS-TFT/Firmware/MKS-TFT_V2.0.0English.zip; use red or blue style 's **retro f**irmware.

makerbase-mks, 2017-06-26:
we have upload at: MKS-TFT/Firmware/MKS-TFT_V2.0.0English.zip; use red or blue style 's  **retro** firmware.

murdock62, 2017-07-23:
Hi, is it possible to have the Status screen in all the style specially in classic please ?

Many thanks !

François.

closed 2017-08-02 by makerbase-mks