makerbase-mks/MKS-TFT #11: no communication

babaz, 2017-02-17:
hello,

i updated the firmware,all is allright,but the screen doesn't communicate with the machine,when i give an order nothing happen.The temperature isn't detected,it doesn't heat,,i setted to autolevel,it's still on manual,It's a makerbase V1.2.

arodriguezf, 2017-02-17:
Try to change the baudrate from the TFT and then restart the board and the
TFT.

babaz, 2017-02-17:
allready done,i tried all the baud rate...mine is 115200...it doesn't work

babaz, 2017-02-24:
same with V2 firmware...

rogerwilco3, 2017-02-26:
Babaz have you looked at the board type ?  before update the firmware i changed the boardtype and baud rate and the ammount of extruders. updated the firmware and working like a charm :-)  hope this gives you a hint where to look

babaz, 2017-02-27:
by board type...you mean mainboard firmware settings ?

rogerwilco3, 2017-03-01:
no the board type in the first line

#########Printer type setting############################
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1 ( I have changed this to 3)

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate setting(1:9600;2:57600;3:115200;4:250000)
>cfg_baud_rate:4 (I have changed this to 3

#language (Simplified Chinese:1; traditional Chinese:2; English:3)
>cfg_language_type:3

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:1

#whether has heated bed(YES:1; NO: 0)
>cfg_custom_bed_flag:1

#--------------------------------------------------------------------
########function settings#################################

than it worked like a charm

Greetings Wilco

burtbick, 2017-03-02:
Hi, I'm trying to get the TFT28 V1.3 working with the MKS Sbase V1.3 board running smoothieware.  I have smoothieware configured for the reprap graphical display on Exp1 and 2 and that works fine.  I have the TFT28 plugged into the appropriate Aux-1 connector.

Updated TFT28 to latest 2.0 FW, and running the retro mix so it has a status display on the main page.
I have my configuration for the TFT28 set just as you have listed above, but I must be missing something on the smoothieware configuration side since the TFT28 is NOT talking to the Sbase board at all.

How did you configure your smoothieware for talking to the TFT28?

Thanks,
Burt

rogerwilco3, 2017-03-02:
hi  Burt,

first update edit the lines  than replace firmware. the firmware and also
the file will have different extension than it should work and directly you
will see it talks to Printer

hope it is working for you

Greetings Wilco

burtbick, 2017-03-02:
Thanks Wilco,  I tried that, doing a FW update again but still no joy getting the TFT28 talking to the SBase V1.3 board.  The configuration file was changed as in your case, and 115200 is the baud rate that is being used between the computer and the SBase board.

Was there anything that you changed on the SBase board Smoothieware configuration to get the TFT28 talking in your case?  Note that I'm running the latest Smoothieware firmware on the MKS SBase board and not the very old version that MKS has on GitHub.

I guess if it continues to not work I'll return the TFT28 and just put OctoPi on a spare RPi and use that with this printer as I'm doing on my PrinterBot Simple Metal.  I was hoping to have the touch screen, but no go so far.

Thanks again,
Burt

burtbick, 2017-03-02:
Even tried reverting to 1.2V of the TFT firmware, but that didn't make a difference either.  Just plain refuses to talk to the SBase board running Smoothieware.  I've asked the vendor of the TFT28 panel for some assistance, but doubt that they will have a good answer, so I'll probably just cut my losses and return the TFT28 via eBay and move on.  Had hight hopes, but it appears that it is a mixed bag as far as people getting it working with the MKS SBase board.  I'll probably try a couple more things for a day or two but if I don't get it talking no sense in holding on to it.

rogerwilco3, 2017-03-02:
Hi Guys Sorry again to be late from work. there is not a lot i changed in
the config of the smoothieboard. As you Burt i do also run the Smoothieware
core instead of the MKS. I would return to the Seller and contact them to
sort this out as there might a bug on your tft screen.

of topic but just a little qeustion have you tryed to hook it also up to a
ethernet cable ?

Greetings Wilco

rogerwilco3, 2017-03-02:
when you want to I can display my full working Config of the Board and TFT
screen.

config file of tft 28 display

#--------------------------------------------------------------
#########Printer type setting############################
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)

>cfg_firmware_type:3

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate setting(1:9600;2:57600;3:115200;4:250000)
>cfg_baud_rate:3

#language (Simplified Chinese:1; traditional Chinese:2; English:3)
>cfg_language_type:3

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:2

#whether has heated bed(YES:1; NO: 0)
>cfg_custom_bed_flag:1

#--------------------------------------------------------------------
########function settings#################################

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

#whether set machine auto-off after print finishes(auto-off:1; NO:0)
>cfg_print_finish_close_Machine:0

#the max target Temp of sprayer and heated bed
>cfg_max_sprayer_temperature:270
>cfg_max_hotbed_temperature:150

#the level signal of outage detection module PB1(low level:0; high level:1)
>cfg_PB0_PB1_Level:1

#the speed of filament replacing(mm/min)
>cfg_filament_change_speed:1200

#the length of filament replacing(mm)
>cfg_filament_change_step:5

#the min temperature of filament replacing
>cfg_filament_limit_temperature:175

#The command of auto leveling (G29 is available for Marlin.While G32 is for
 Repetier and Smoothieware)
>cfg_auto_leveling_cmd:G28;G29;

#The point number of manual leveling:(3,4,5 point available)
>cfg_point_number:4

#the coordinates of 5 point on manual leveling
>cfg_point1:50,50
>cfg_point2:180,50
>cfg_point3:180,180
>cfg_point4:50,180
>cfg_point5:150,150

#the moving speed of leveling(mm/min)
>cfg_leveling_z_speed:1500
>cfg_leveling_xy_speed:3000
[... 584 lines cut ...]
return_error_on_unhandled_gcode              false            #

# network settings
network.enable                               true            # enable the
ethernet network services
network.webserver.enable                     true             # enable the
webserver
network.telnet.enable                        true             # enable the
telnet server
#network.ip_address                           auto             # use dhcp
to get ip address
# uncomment the 3 below to manually setup ip address
network.ip_address                           192.168.1.221    # the IP
address
network.ip_mask                              255.255.255.0    # the ip mask
network.ip_gateway                           192.168.1.1      # the gateway
address
#network.mac_override                         xx.xx.xx.xx.xx.xx  # override
the mac address, only do this if you have a conflict

hope this will clear your issue

Greetings Wilco

2017-03-02 19:17 GMT+01:00 wilco westende <rogerwilco3@gmail.com>:

burtbick, 2017-03-02:
Thanks, Other than the number of extruders my TFT config is the same.  And from what I can see skimming your Smoothie config I've got the same settings (the ones that would be potentially critical) as you.  I'll do a more detailed compare a little later just in case I missed something.

I was coming to the conclusion that there may be a defect with this TFT28 board that is preventing it from talking to the SBase board.

This is where it would be nice if MKS would release the schematics for their board, as they really should since it is derived from the Open Source Smoothie board.  But that is probably too much to hope for.  I like a number of the features on the MKS board but probably will purchase a real Smoothie board for my other machine(s) in the future.

I'll let you know if I find anything as a solution for others.

I have tried the SBase board via Ethernet and that works, but the interface is pretty minimal.  If I can't easily resolve the TFT28 I'll probably just go for the OctoPi solution, which won't be too bad since  already have the RepRap graphical display anyway.

Thanks again,

burtbick, 2017-03-02:
OK, I fixed it.  The problem was the supplied cable for connecting the TFT28 to the controller board!  It appears to have a bad crimp so that was why it wasn't talking to the SBase board.  Just updated back to V2.0 firmware to get the status display on the main screen, and it is still working so it appears that it was a bad cable that was the root cause of the problem, at least in my case.

Thanks for all your help..

Burt

babaz, 2017-03-03:
i've done all what you said but it doesn't work

burtbick, 2017-03-03:
babaz, are you using an MKS SBase V1.2 or newer main board?  Also, be sure that you have the ribbon cable plugged into Aux 1 on the SBase board AND also that it is plugged into the serial I/O connector on the TFT panel.  That is the one that as tx rx GND and +5V next to it.  There are 3 identical connectors on my TFT panel.  I'm sure you have the cable plugged into the correct one on the panel, but it might be worth double checking.

And, finally, in my case the problem turned out to be a defective ribbon cable.

To recap:

I set my TFT config file up with the board type as smoothie as has been described above, and the baud rate as 115200 and english for the language.  Note I started with the V2.x config file and made changes there.

I started fresh with an empty SD card (removed the old file and saved just in case).

I downloaded the latest V2.x of the firmware from the github repository.

Extracted the files, copied the following from the MKS-TFT_V2.0.0English/blue_style/Retro directory:
mks_pic
mksft28.bin
And as mentioned I edited the mks_config.txt file from that directory making the changes as described above.

Unmounted the SD card and popped it back into the TFT28.  Powered up by applying 12V power to the MKS SBase board and after the firmware and mks_pic files were installed I was able to control steppers, fan, and see the temperature from the hot end and the heated bed (well in my test case I have a spare hot end connected sitting on the bench and an extra thermistor connected as the hot bed one, and the extra steppers sitting on the bench with the spare cooling fan connected.

I haven't connected everything to my Tronxy802 yet, that will be happening in a few days after some more testing.

But in any case after I replaced the ribbon cable with a spare I had laying around in the junk box I was able to talk to the MKS SBase board with no problems.

So double check where you have the ribbon cable plugged in on the TFT board, just in case.

If you have a spare ribbon cable laying around try changing that out.  If you don't and you have an ohms meter or continuity tester try ringing out the ribbon cable to make sure all the pins are connected.

Hope that helps.

babaz, 2017-03-03:
base V1.2

burtbick, 2017-03-03:
OK, then I don't have any suggestions other than double check your ribbon cable as I mentioned, both to make sure it is connected to the correct connector on the TFT panel, and that there are no breaks or shorts in the cable itself.

In my case I have the 'S'Base V1.3 board that is running Smoothieware not just the Base board.  Usually pre-made ribbons are good, so I don't know in your case if that is likely to be the culprit.  I was able to check mine with the meter ringing it out pin for pin, using a couple of resistor leads to plug into each pin on the connectors, then check for continuity from pin 1 to pin 1 and so on.  Shorts between cables are less likely.

java402000, 2017-03-16:
Hi I got the mks 1.5 board and the touch screen 3.2. Evey thing seems to work.  I can move all axis I can pre heat and home all axis. and i can print from computer but stand alone cant print. I go to print from the usb, I see the g-code ,it lets me enter the screen goes to what looks like a print screen then the printer just sits there. Any ideals whats wrong .﻿

rogerwilco3, 2017-03-18:
Hello Java,

are the thermistors detecting the right temperature ?
and you can see it is warming up ?
if it is possible in your computer Change the value T to 19 make sure you
do same for the heated bed. remove Fillament and do a test run to see if it
is working than for you.

Greetings Wilco

java402000, 2017-03-19:
when I pre heat  it detects the temp, I dont think its right 55 c n bed i still can hold my hand on it. I conldn't with the Melzi board.  When printing with computer hook to it, I had to turn up the heat on both the bed and hot end. I was printing pla. On changing the t value, I'm lost, I assume that you are talking about the firmware. The only stuff i work with there is in config h and adv config h. I went through them and didn't see what you meant. I did see that it has me set
  #define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_BED 1 Sorry about bing a "New bee" i am learning a lot.
Thanks for everyone's help
Tom

rogerwilco3, 2017-03-19:
Hey Tom,

Well first of all we all learning so dont worrie

the config i was talking about is the 32bits version on the board.
On Configuration.h you run the 8bit variant.
Well in that case in front of every line where a // is will be ignored
you use in your settings as they are now 1 hotend and 1 heatedbed.
But what type of thermistor are you using ?  You can set the value of the
parameter to 0 (not used), 1 (RRW thermistor 100K) or -1 (RRW thermo
couple).
But even if you are not warming up the bed or extruder it should a reading
of outside tempurature allready.
#define TEMP_SENSOR_0 1  (used and selected and using the rrw thermistor
100k)
#define TEMP_SENSOR_1 0 (not used)
#define TEMP_SENSOR_2 0 (not used)
#define TEMP_SENSOR_BED 1 (used and selected and using the rrw thermistor
100k)

Also make sure your boardtype is set

Greetings Wilco

rogerwilco3, 2017-03-19:
hi Tom,

Some more information regarding Thermistors etc.

//=============================Thermal Settings
 ============================
//===========================================================================
//
//--NORMAL IS 4.7kohm PULLUP!-- 1kohm pullup can be used on hotend sensor,
using correct resistor and table
//
//// Temperature sensor settings:
// -2 is thermocouple with MAX6675 (only for sensor 0)
// -1 is thermocouple with AD595
// 0 is not used
// 1 is 100k thermistor - best choice for EPCOS 100k (4.7k pullup)
// 2 is 200k thermistor - ATC Semitec 204GT-2 (4.7k pullup)
// 3 is Mendel-parts thermistor (4.7k pullup)
// 4 is 10k thermistor !! do not use it for a hotend. It gives bad
resolution at high temp. !!
// 5 is 100K thermistor - ATC Semitec 104GT-2 (Used in ParCan & J-Head)
(4.7k pullup)
// 6 is 100k EPCOS - Not as accurate as table 1 (created using a fluke
thermocouple) (4.7k pullup)
// 7 is 100k Honeywell thermistor 135-104LAG-J01 (4.7k pullup)
// 71 is 100k Honeywell thermistor 135-104LAF-J01 (4.7k pullup)
// 8 is 100k 0603 SMD Vishay NTCS0603E3104FXT (4.7k pullup)
// 9 is 100k GE Sensing AL03006-58.2K-97-G1 (4.7k pullup)
// 10 is 100k RS thermistor 198-961 (4.7k pullup)
// 11 is 100k beta 3950 1% thermistor (4.7k pullup)
// 12 is 100k 0603 SMD Vishay NTCS0603E3104FXT (4.7k pullup) (calibrated
for Makibox hot bed)
// 13 is 100k Hisens 3950  1% up to 300Â°C for hotend "Simple ONE " &
"Hotend "All In ONE"
// 20 is the PT100 circuit found in the Ultimainboard V2.x
// 60 is 100k Maker's Tool Works Kapton Bed Thermistor beta=3950
//
//    1k ohm pullup tables - This is not normal, you would have to have
changed out your 4.7k for 1k
//                          (but gives greater accuracy and more stable PID)
// 51 is 100k thermistor - EPCOS (1k pullup)
// 52 is 200k thermistor - ATC Semitec 204GT-2 (1k pullup)
// 55 is 100k thermistor - ATC Semitec 104GT-2 (Used in ParCan & J-Head)
(1k pullup)
//
// 1047 is Pt1000 with 4k7 pullup
// 1010 is Pt1000 with 1k pullup (non standard)
// 147 is Pt100 with 4k7 pullup
// 110 is Pt100 with 1k pullup (non standard)

#define TEMP_SENSOR_0 3
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_BED 3

// This makes temp sensor 1 a redundant sensor for sensor 0. If the
temperatures difference between these sensors is to high the print will be
aborted.
//#define TEMP_SENSOR_1_AS_REDUNDANT
#define MAX_REDUNDANT_TEMP_SENSOR_DIFF 10

2017-03-19 11:17 GMT+01:00 wilco westende <rogerwilco3@gmail.com>:

closed 2017-06-27 by makerbase-mks