makerbase-mks/MKS-TFT #12: Status screen all messed up on TFT32

Bigfellas, 2017-02-23:
I am running an MKS TFT32 The status screen is not displaying properly. All other menus and buttons are fine. Any thoughts. see picture
https://1drv.ms/i/s!AgFOUJh3Fjh1gqQUE5e9Aa_heQHJag

makerbase-mks, 2017-02-24:
sorry,i can't see the picture.
can you send email to king@makerbase.com.cn

Bigfellas, 2017-02-24:
Thanks. I have sent that to your email.

pmbroth, 2017-03-03:
I had that happen as well, switch to the retro firmware, you wont have that happen,  I forgot which one it was either simple or the other one, I cant remember the third one.  But I know 2 our of the three works, and Retro definitively worked.

makerbase-mks, 2017-03-04:
Hello,is your MKS TFT V1.0 version? This theme just can be used above V 1.1 version.

Bigfellas, 2017-03-04:
Yes it is a v1.0. So sad for me

closed 2017-03-31 by makerbase-mks