makerbase-mks/MKS-TFT #13: Do I miss understand what filament change

Bigfellas, 2017-02-24:
I was thinking filament change was for loading and unloading filament. But no matter what number I enter in 
#the speed of filament replacing(mm/min)
>cfg_filament_change_speed:1200

#the length of filament replacing(mm)
>cfg_filament_change_step:1

it loads and unloads well over a meter of filament.
am I using this the wrong way.

mks67, 2017-03-05:
I have got the same problem.

changing the length of filament doesn`t make any difference at all.

makerbase-mks, 2017-04-01:
Hello dear,there are 15 commands for filament IN/OUT,so it will not stop until 15 commands finish.
Or you are able to try  slower speed and  shorten length of filament .

#the speed of filament replacing(mm/min)
>cfg_filament_change_speed:500

#the length of filament replacing(mm)
>cfg_filament_change_step:1

makerbase-mks, 2017-06-27:
this function is not good enough, we will change it in V3.0

closed 2017-08-02 by makerbase-mks