makerbase-mks/MKS-TFT #14: fan icon not moving or showing speed

pmbroth, 2017-03-01:
Hello,  I noticed if I change the speed via the controller the speed is noted on the status screen and the shows movement.  When I print via Simplify3d, Cura or Mattercontrol the fan will be on, but the status screen does not show the speed or movement.

salcofago, 2017-03-06:
Ii also encounter this problems , as well the text in some of the icons after you start the print is in Chinese

pmbroth, 2017-03-31:
I tried the latest software which was uploaded a few days ago, still has the fan issue with  not show movement or record speed if not controlled on the controller, if software is used to turn on the fan no speed indication.

makerbase-mks, 2017-04-01:
Hello dear,If use pronterface online printing,there is no feedback and TFT doesn't show rotate speed detail.

Wrong icon problem had been fixed,and we updated new firmware already.

pmbroth, 2017-04-02:
I used the latest firmware, still the same issue with- curator, simplify3d, matter control.  Fan image and speed does not work.  Only work when you turn on by the controller.

makerbase-mks, 2017-06-26:
MKS TFT use M105 to get the board's status, so TFT can only  get temperature status. When use cura change Fan speed, MKS TFT can not get the new Fan status. So this problem can't be solve because there has not  M-commands like M105 to get Fan status.

closed 2017-06-27 by makerbase-mks

SmoothBro, 2017-06-28:
Hi makerbase-mks, is there any way to get feedback from the tft to pronterface? is it possible to edit the firmware or make a workaround to get these reports such as M114, M117, M119? I feel like these commands can benefit the user experience.

If not, would it be possible to also put the X axis position and Y axis position next to the Z axis that you've added at the top of the screen in the move menu. I think its a good idea to know where the extruder is at at all times.

Also in the move axis menu if there was an extra +100mm option these features would pretty much perfect the mks tft experience.

I look to purchase these items in bulk for my product so please let me know if you can add any features.