makerbase-mks/MKS-TFT #15: No way to disable leveling menu altogether?

jfgomez86, 2017-03-09:
**I'd like to disable the "Leveling" menu so that it doesn't show up on the screen, or at least let me edit the functions of the "Leveling" menu buttons so that I can match with my calibration routine.**

I wanted to be able to be able to configure Z offset on my delta on the same menu I performed my calibration. In order to achieve this, I used the "More" custom button entries to add these (Smoothie):

```
# bmp_custom1-7.bin
>moreitem_button1_cmd:G28; M280 S3.0; G32 I0.02; M280 S7.0; M500; G28; M280 S3.0; G31; M374; M375; M500; G28; M280 S7.0; G1 Z50;
>moreitem_button2_cmd:M280 S3.0; G1 X0 Y0 Z8 F4000; G30 Z1.05; M280 S7.0;
>moreitem_button3_cmd:G91;G1 Z0.1; G90;
>moreitem_button4_cmd:M84;
>moreitem_button5_cmd:M306 Z0; M500;
>moreitem_button6_cmd:G28;
>moreitem_button7_cmd:G91;G1 Z-0.1; G90;
```

*button1*: Auto-Calibrates (G32 I0.02) and Performs Grid Compensation Routine (G31). Saves results.
*button2*: Approaches the hotend to the bed (using probe first to avoid crashing against bed). This is to proceed with Z-offset using *button5*.
*button3*: Manually raises hotend by 0.1mm to help set Z-offset.
*button4*: Shuts off steppers.
*button5*: Sets Z Offset (using M306) and saves to EEPROM.
*button6*: Home Axis (G28).
*button7*: Manually lowers hotend by 0.1mm to help set Z-offset.

This is more like the menu I'd like to see for Auto-Calibrating a Delta.

makerbase-mks, 2017-04-01:
Dear Sir,thank you so much for your advise,we will improve it in the future.

makerbase-mks, 2017-06-27:
you can disable  "Leveling" menu in V3.0, it will be release before 7.22. thank you

closed 2017-08-02 by makerbase-mks