makerbase-mks/MKS-TFT #19: Does not respond to M300 or M600

veedubb65, 2017-03-31:
M300 in a gcode file should cause the buzzer to beep. It has no effect.

M600 in a gcode file should cause the printer to pause, mid print, for a filament change. It should also show appropriate tools on the screen for changing filament. It has no effect.

MKS TFT28, MKS firmware 2.0. Latest Marlin Firmware on MKS Base board.

makerbase-mks, 2017-04-01:
Hello dear,I am sorry that mianboard does not support M300 to run buzzer.
there are 15 commands for filament IN/OUT,so it will not stop until 15 commands finish.

closed 2017-08-02 by makerbase-mks

pauloblank, 2019-04-29:
The board has got a buzzer but no way to use it? If M300 does not work, should be another way of using the buzzer. Please correct this.