makerbase-mks/MKS-TFT #24: Error messages?

cruwaller, 2017-04-10:
Is there way to see error messages? Now the GUI is not very informative...

TFT32 with 2.0 firmware has a printing issue and stops suddenly at some point, GUI get stuck (unresponsive) but no information why!!
How to debug the issue??
TFT32 is connected to the SBASE v1.3 which works great (all SD, LAN and USB printing is ok).

Currently the display is useless.

makerbase-mks, 2017-06-27:
Because this function need higher speed MCU. We can't add this to TFT. Sorry.

closed 2017-06-27 by makerbase-mks

cruwaller, 2017-06-27:
Haha, really? You are using 32bit ARM processor and there is not CPU time to process error messages from uart? Wtf!
How can you process / feed gcode to printer controller? 
Currently this LCD is useless since i cannot locate the error/what happened...

Strange...