makerbase-mks/MKS-TFT #26: Version for MKS-TFT for SBASE running a Laser

mayhem2408, 2017-04-18:
Are there any plans to make a firmware for the TFT more suited to laser CNCs instead of 3D Printers. I have just purchases a MKS SBASE to upgrade my K40 Laser and the colour touch screen would add a nice look to it.

benmprojects, 2017-05-17:
I would also like to see this.. Please advise if you are considering doing this. Im sure people will donate to fund this.

makerbase-mks, 2017-06-27:
good idea, but we will do it after 2-3 month. thanks.

closed 2017-06-27 by makerbase-mks

cmenchion, 2017-12-05:
Any progress on this? Desperately need this.

Ruisjm1, 2018-05-31:
Is possible create CNC interface for jogging and sending files from sd card? and connect in rx and tx arduino running GRBL to make a stand alone controller?

I think people will love it and buy a lot MKS TFT