makerbase-mks/MKS-TFT #30: pausing when print from sd/usb

handsomedude, 2017-05-22:
every now and again the printer pauses printing from sd/usb device. waits for temp to reach goal again and carries on.  causes a blob of plastic on print.

i dont get this printing direct from pc.

mkstft 28 version 2.0

csikiati, 2017-05-29:
check in config below value of EXT0_HEAT_MANAGER to be 1. anyway, I am surprised your printer even works, I don't have settings related to steps/mm or what is below in config file so I have decided to use the simple MKS robin config file, which has these options. What is your secret to make it work?

#Type of heat manager for this extruder.
#- 0 = Simply switch on/off if temperature is reached.
#- 1 = PID Temperature control.
#- 3 = Dead-time control.

>EXT0_HEAT_MANAGER	1

makerbase-mks, 2017-07-29:
Hello,The plastic on the printer is due to the high temperature of the printer, which makes the consumables melt.

closed 2017-08-02 by makerbase-mks