makerbase-mks/MKS-TFT #38: Repetier support

nic6911, 2017-06-28:
Hi!
I have an MKS TFT28 talking to a board running repetier firmware. I can do manual control, heat up nozzle and buildplate, home etc. But as soon as I load a file to print it does not work. Somehow it seems like communication breaks down. Temperatures stay the same, nothing moves, only the timer counts up on the TFT28. I can cancel the print and move around in the menus, but after starting a print (and canceling it again) I can not do manual control any longer. 
I have to restart the machine to do manual control from the TFT28 again.

The mainboard on the printer is still controlling temperature when the TFT28 communication crashes.

In the config I have chosen repetier as the firmware. I have also tried the two other options, just to see whether it could help. I have tried with several versions of the firmware, and also the new 2.0.1.

makerbase-mks, 2017-07-31:
You can try to replace the SD card.

nic6911, 2017-07-31:
The card I use works fine on other printers. But I just tried with a new SD-card (as in brand new), same result. It does the same when printing from USB stick. I am able to navigate in the SD-card folder from the TFT28 menu, but when I press a gcode to stat printing, the communication to the printer control board seems to break down. I can still go around in the menus of the TFT28, but the communication to and from the TFT28 is not working. As mentioned in the prior, it works again if I remove the power and plug it back in again. 
So my only option for printing is through the USB of the printer control board at the moment.

Sicon76, 2017-07-31:
This is the same thing we are discussing in another topic about pause on printing from SD, I see it happens with repetier firmware too other than Marlin and derived. I know MKS are working on this issue, hope they will find a solution soon for every kind of firmware (because the LCD is cheap and really good other than this problem)

closed 2020-03-12 by makerbase-mks