makerbase-mks/MKS-TFT #39: Option for 4 extruders?

Tertius91820, 2017-07-19:
Good day

is there a option to add 4 extruders on the mks tft 28?

I'm using the kraken extruder which uses 4 hot ends.

thank you

makerbase-mks, 2017-07-29:
Hello! Our MKS TFT28 can only show two extruded headers at most, and if you want to show four extruded heads, our company can customize the products for you.Hope this answer is helpful to you, I wish you a happy life!

cruwaller, 2017-07-29:
Just release it for everyone. Why to make just a custom release?

makerbase-mks, 2017-07-29:
We don't have a custom version of our company now, but we will develop it if the customer needs it.

closed 2017-07-31 by makerbase-mks

Tertius91820, 2017-08-05:
I need 4 extruders!

neverna-ptitsa, 2018-03-11:
+1 for multiple (4) extruders - this will allow printing with Kraken. Some reasons to do this might be incorporating support material, metal material, and ceramic material, for example. Or support material, hard thermoplastic, and soft thermoplastic.

rikvandersteen, 2021-01-18:
Hello is there already a version for 4 extruders    I'm looking for the 7 inch touch screen from makerbase version for my zonestar z9m4   that has 4 extruders