makerbase-mks/MKS-TFT #41: Status screen not working

copystring, 2017-07-27:
I'm using a MKS SBASE V1.3 with MKS TFT32 V1.1
Any idea what is wrong? Same issue as here: https://github.com/makerbase-mks/MKS-TFT/issues/12
But I have that V1.1 display and it's still not working.

[image]
[image]

`#--------------------------------------------------------------
#########Printer type setting############################
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:3

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate setting(1:9600;2:57600;3:115200;4:250000)
>cfg_baud_rate:3

#language (Simplified Chinese:1; traditional Chinese:2; English:3)
>cfg_language_type:3

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:1

#whether has heated bed(YES:1; NO: 0)
>cfg_custom_bed_flag:1

#--------------------------------------------------------------------
########function settings#################################

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

#whether set machine auto-off after print finishes(auto-off:1; NO:0)
>cfg_print_finish_close_Machine:0

#the max target Temp of sprayer and heated bed
>cfg_max_sprayer_temperature:270
>cfg_max_hotbed_temperature:150

#the level signal of outage detection module PB1(low level:0; high level:1)
>cfg_PB0_PB1_Level:1

#the speed of filament replacing(mm/min)
>cfg_filament_change_speed:1200

#the length of filament replacing(mm)
>cfg_filament_change_step:5

#the min temperature of filament replacing
>cfg_filament_limit_temperature:175

#The command of auto leveling (G29 is available for Marlin.While G32 is for  Repetier and Smoothieware)
>cfg_auto_leveling_cmd:G28;G32;

#The point number of manual leveling:(3,4,5 point available)
>cfg_point_number:4

#the coordinates of 5 point on manual leveling
>cfg_point1:50,50
>cfg_point2:180,50
>cfg_point3:180,180
>cfg_point4:50,180
>cfg_point5:150,150

#the moving speed of leveling(mm/min)
>cfg_leveling_z_speed:1500
>cfg_leveling_xy_speed:3000

#Breakpoints continued playing Z-axis error setting
>cfg_breakpoint_z_error:0.2

#wifi model(0:MKS WIFI TFT  1:MKS HLK-WIFI)
>cfg_wifi_type:1

#wifi mode(AP:1; STA:2)
>cfg_wifi_mode:2

#wifi name and password
>cfg_wifi_ap_name:
>cfg_wifi_key_code:

#0:disable cloud service 1:enable cloud service
>cfg_cloud_enable:0
#cloud service adress
>cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#IP/mask/gateway
>cfg_ip_address:192.168.3.100
>cfg_ip_mask:255.255.255.0
>cfg_ip_gate:192.168.3.1

#------------------------------------------------

##############Function customization####################

#user-defined function1 and 2.(whether display this button .  disable: 0; enable: 1)
>cfg_function_btn1_display:1
>cfg_function_btn2_display:0

#command of user-defined function
>function_btn1_cmd:M84;
>function_btn2_cmd:M81;

#set number of  "More" button 
>moreitem_pic_cnt:0

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G28 X0;

>moreitem_button2_cmd:G28 X0;

>moreitem_button3_cmd:G28 Y0;

>moreitem_button4_cmd:G28 Y0;

>moreitem_button5_cmd:G28 Z0;

>moreitem_button6_cmd:G28 Z0;

>moreitem_button7_cmd:G28;

#how many "More" button show on screen interface
>morefunc_cnt:0

#command setting on button1~6
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#---------------------------------------------------
##Display customization(button and logo customized see User's Manual)####

#the color of screen background
>cfg_background_color:0x494949

#the color of title text
>cfg_title_color:0xFFFFFF

#the color of temperature,fan,(etc) background
>cfg_state_background_color:0x494949

#the color of temprature,fan,(etc) text
>cfg_state_text_color:0xFFFFFF

#the color of file button
>cfg_filename_background_color:0x494949

#the color of file text
>cfg_filename_color:0xFFFFFF

#the color of background in printing 
>cfg_printingstat_word_background_color:0x494949

#the color of text in printing status
>cfg_printingstat_word_color:0xFFFFFF

#3D effects setting (disable: 0; enable: 1)
>cfg_BUTTON_3D_effects:0`

makerbase-mks, 2017-07-29:
Hello! The reason for this phenomenon is that the firmware does not support the MKS TFT32 V1.1 version. Please use the hardware of the MKS TFT32 V2.0 version. I hope it can be helpful to you and wish you a happy life!

closed 2017-07-29 by copystring

undertowx, 2017-11-04:
My mks tft 32 won't display real time ext or bed temps. It is running with a mks s base v1.3. Please help