makerbase-mks/MKS-TFT #42: Retro in Version 2.0.1 32english not work

GaryK47, 2017-08-01:
Hello!
It will only update the images but not the firmware. The file name is incorrect. If you change the filename from mkstft24.bin to mkstft28.bin, the upload works and the display only shows "Booting ...".

makerbase-mks, 2017-08-05:
Are you used hardware of MKS TFT24?

GaryK47, 2017-08-05:
I use MKS TFT3.2_L V.2.0.

ovsel43, 2017-08-05:
On MKS TFT 28 version 2.0 also does not work.
When updating the display, the process reaches 100% and hangs. After rebooting the display hangs on the "booting..."

ovsel43, 2017-08-05:
Please upload English firmware 2.0.1 retro for tft 28 and 32.

ovsel43, 2017-08-05:
Retro firmware from the archive "TFT28 32 简体 中文 V2.0.1.zip" works fine on my TFT.
In "TFT 28 32English2.0.1.zip" the retro firmware is not for that display.

makerbase-mks, 2017-08-07:
We will updat it in today.

makerbase-mks, 2017-08-07:
We have updated it.

closed 2017-09-08 by makerbase-mks