makerbase-mks/MKS-TFT #43: mks 2.8 usb odn't read

patacco, 2017-08-06:
hi, i've tried with many usb sticks but it seems it not function. i've choosen usb in filesystem menu , but it seems it not read ... any suggestion?

makerbase-mks, 2017-08-07:
there is the file system in the option of set.do you set it of udisk?

nemocaledonie, 2017-08-13:
As Patacco but on a 3.2 TFT, the screen can't read my udisk, and I already do the change on SET.FILES.
I'm using the latest firmware 2.0.1 English and Marlin 1.1.x

makerbase-mks, 2017-08-14:
Some usb drives are unsupported and we suggest replacing the udisk.

nemocaledonie, 2017-08-14:
Ok I'll try another udisk. Thanks for the reply

kiar8, 2017-09-06:
I apologize for my english
I have a tft 3.2 display with the latest version of firmware downloaded from this site and marlinkimbra works everything but when i need to print from sd or usb card it does not work

closed 2020-03-12 by makerbase-mks