makerbase-mks/MKS-TFT #44: MKS 2.8 Not booting after flash

sylus28, 2017-08-07:
Hi my TFT2.8 still block booting step
Is there a way to fix?

makerbase-mks, 2017-08-07:
What the version of your firmware?

makerbase-mks, 2017-08-07:
What is the version of your firmware?

sylus28, 2017-08-07:
2.0.1 latest

sylus28, 2017-08-07:
reflash with latest and it works again

makerbase-mks, 2017-08-08:
#29 
![Mou icon](https://image.baidu.com/search/detail?ct=503316480&z=0&ipn=d&word=%E5%A4%9A%E8%82%89%E6%A4%8D%E7%89%A9&step_word=&hs=3&pn=8&spn=0&di=22559192280&pi=0&rn=1&tn=baiduimagedetail&is=0%2C0&istype=2&ie=utf-8&oe=utf-8&in=&cl=2&lm=-1&st=-1&cs=3311295539%2C1946101578&os=3350954942%2C28771632&simid=0%2C0&adpicid=0&lpn=0&ln=1976&fr=&fmq=1502159259948_R&fm=&ic=0&s=undefined&se=&sme=&tab=0&width=&height=&face=undefined&ist=&jit=&cg=&bdtype=10&oriquery=&objurl=http%3A%2F%2Fwww.cd-pa.com%2Fbbs%2Fdata%2Fattachment%2Fforum%2F201606%2F08%2F202531w8zvcfpfweefve29.jpg&fromurl=ippr_z2C%24qAzdH3FAzdH3Fooo_z%26e3Bv1-rw_z%26e3Bv54AzdH3FkkfAzdH3Fpi6jw1-nabbb0c-8-8_z%26e3Bip4s&gsm=0&rpstart=0&rpnum=0)

makerbase-mks, 2017-08-08:
[image]

closed 2017-08-08 by sylus28