makerbase-mks/MKS-TFT #51: Request: option to set pre-heat temperatures in mks_config.txt

3DFreezeMe, 2017-08-29:
Hi,

It would be a nice feature when we could set custom values for the pre-heat temperatures in the mks_config.txt file. When going to the heating menu at this moment, pre-heating starts with default values which I have to change manually each time.

Even better would be if we could have 3 pre-heat options available to configure (either via separate buttons or a selection icon) available to set ourselves. PLA, ABS and PETG option would be great, or better set names in the mks_config.txt file.

Thanks!

makerbase-mks, 2017-08-30:
Thanks for your advice.If you want to do this, you can design the icons to replace icons of 'More' and configure instructions in mks_config.txt file.

makerbase-mks, 2017-08-30:
Thanks for your advice . If you want to do this , you can design the icons to replace icons of 'More ' and configure instructions in mks_config . Txt file .

3DFreezeMe, 2017-08-30:
Yes, I understand that I could use the 'More'  button options,  but why not have the default option configurable? That should not be to difficult to add to the, sadly still closed, firmware?
Now I have to 'waste' 3 'More' buttons to achieve something that, in my opinion, should be user configurable.

makerbase-mks, 2017-08-30:
Thanks for your advice . If you want to do this , you can design the icons to replace icons of 'More ' and configure instructions in mks_config . txt file .

3DFreezeMe, 2017-08-30:
Wow, do you use an answering bot now?

makerbase-mks, 2017-08-31:
It is not an answering bot.For your suggestion , we will consider it in the future work . If you need this feature now , you can make a customized version .

closed 2020-03-12 by makerbase-mks