makerbase-mks/MKS-TFT #52: How Work the leveling process ?

Gromak123, 2017-08-30:
Hi ! 

I have perfectly confogured my MKS Sbase motherboard and I just add a TFT28 to the setup, all is working fine but I don't understand the leveling process.. when I select one of the five point my Extruder Go High on Z and never stop .. 

Could you help me ? 

Ps : The homing Work perfectly fine.

Gromak123, 2017-08-31:
Can you tell me how work the leveling process ? I can't level my bed because this functionality doesn't work.

makerbase-mks, 2017-09-01:
You can download the files from the website below , finding your answer in "Zprobe.htm"  file if the data sheet can help you .
https://github.com/makerbase-mks/MKS-SBASE/blob/master/English/help.zip

closed 2020-03-12 by makerbase-mks