makerbase-mks/MKS-TFT #53: Mks_WIFI.ino.bin ??? and Changing filament button

nemocaledonie, 2017-08-31:
Hi, In the directory /Red style/Classic/ you have a file (and only in this folder) Mks_WIFI.ino.bin, so when I'm updated my TFT,  I copied this file too in the SD card, except now the WIFI don't working any more.
How I can go back to the original configuration ? (i've still "EXCEPTION" in the WIFI menu).

Second question : how you can change the filament when you printing ? no button or something like that. Only when the printer don't work (It's not very helpful, because you can do it manually and faster)

makerbase-mks, 2017-09-01:
what is the version of your firmware?

nemocaledonie, 2017-09-01:
Ver. 2.0.1 English.

makerbase-mks, 2017-09-02:
We're testing it here and it's working. Are you sure your phone is connected to "MKS_wifi"?

nemocaledonie, 2017-09-02:
Of course ! and the password "makerbase"

makerbase-mks, 2017-09-02:
We have released the latest firmware V3.0.0, you can  download and update it.

cruwaller, 2017-09-02:
What's new in 3.0? Bug fixes? Readme is not updated...

nemocaledonie, 2017-09-03:
Thanks for the 3.0.0. Now I can change the filament (no trying yet but the button is on the screen), and the "exception" has gone...

closed 2017-09-03 by nemocaledonie