makerbase-mks/MKS-TFT #54: TFT32 Not working

murdock62, 2017-09-01:
Hello,
I encounter a problem with the screen.
When I connect it it just beep beep, beep, beep, beep beep ... But the screen remains black.
To eliminate a connection problem I replaced it with the screen of one of my other machine and everything works.
Could you help me please?
Many thanks !
Best regards.
François.

makerbase-mks, 2017-09-01:
what is the version of your firmware and hardware?

murdock62, 2017-09-01:
Hardware version 2.0. Firmware dont know, the screen never work.

Many thanks for your help.

makerbase-mks, 2017-09-02:
Doesn't the screen show "booting"?

murdock62, 2017-09-02:
Nope. Anything on the screen.
Only continuous beep beep.

makerbase-mks, 2017-09-04:
Where your stuff is bought, you can contact customer service and they will help you.

mks83, 2018-06-01:
Same exact problem here, can somebody please post an outcome

makerbase-mks, 2018-06-11:
Maybe you should contact sales staff to replace this screen.

closed 2020-03-12 by makerbase-mks