makerbase-mks/MKS-TFT #56: printing from usb or sd

kiar8, 2017-09-06:
I apologize for my english
I have a tft 3.2 display with the latest version of firmware downloaded from this site and marlinkimbra works everything but when i need to print from sd or usb card it does not work

Sicon76, 2017-09-06:
I and many other (I am still searching for owners of this tft that can print from sd using a recent version of Marlin or Marlinkimbra) have the same problem (there is even a thread in this section), but still can't find a solution. MKS said it was our firmwares, but maybe it's better to solve this problem from mks-side. By now we ar eusing it for everything but print from sd, and it is a pity. Hope they will find a solution soon

kiar8, 2017-09-06:
I hope I find it in a short time

makerbase-mks, 2017-09-12:
I've configured a firmware of Marlinkimbra. I can print it from SD card on my machine. You can download it and try to print from SD card.
https://github.com/makerbase-mks/Marlin/blob/master/Marlinkimbra_15-07-2017_0351am_delta.rar

Sicon76, 2017-09-12:
As I suggested to you before, the version you used is an old and discontinued version of marlinkimbra, the new one is the MK Duo, you should take it in this repository https://github.com/MKFirmware/MK4duo  . If you can try with that and works than please tell us which options you used because we are so many waiting to fix this issue.

Thank You!

kiar8, 2017-09-12:
I did a quick test with the linked version but it does not work for me yet if others succeed in trying this better

nemocaledonie, 2017-09-14:
For me the printing from the SD or USB working properly, and I'm using the latest firmware from Marlin.

kiar8, 2017-09-14:
I have tried with marlin 1.1.5 and tft 32 display with firmware version 30
and i can not run you can turn a copy of your marlin so i do the tests
thank you

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

nemocaledonie, 2017-09-21:
Hi every all,
So I made some test with the new Marlin (28/08/17) and I 'm thinking the TFT have an incompatibility with the NEOPIXEL library used in this new Marlin. Why I say that ? because I modified the previous Marlin for working with my new NEOPIXEL strip, (previously I'm tryed with the new Marlin without success), and I modified the old Marlin, but only with adding the NEOPIXEL library (and some code of course :-) ), and nothing working : TFT, PWC (the TFTWIFI still don't worked but don't worked before).
So the NEW MARLIN DON'T WORK WITH THE TFT ! 
(or I'm pretty sure... yes, I know we can't be sure for anything except the death ...)

jorgehto, 2018-10-04:
Hi All was this solved? I have a MKS GEN V1.4 with last Marlin Version and a MKS TFT32 Full Graphic Smart Controller with last firmware on it, and the SD printing is not working...

robustini, 2018-12-03:
Same problem here with MarlinKimbra V4.3.7, all wotk fine from the display but not the print from USB or SD.