makerbase-mks/MKS-TFT #59: Tft28, pwc and ramps

Torro666, 2017-09-26:
Hi there,

I can't get the pwc to work.
The printer starts as long as i push the Button. Button released, printer off.
there is no connection from pwc to ramps.

Does the pwc even work with ramps 1.4 or do i need a sbase?
I thought the printer off command works with the tft and a board is Not needed to shut the printer off after finish print.

Thanks and Best regards,

Florian

nemocaledonie, 2017-09-26:
Hi Florian,
I had the same problem but now, everything work. Normally it's working on RAMPS (the MKS Base is nearly the same board), you must affected a pin in Marlin, using the right pin on TFT and connecting the blue/green/yellow on the PWC in the good order. 
So :for the wires : the right way is : blue-green-yellow, for the board choose D11 or D12 on the board and in Marlin do the same (D11 or D12 in pin.h), put the POWER_SUPPLY on "2" in configuration.h. In TFT choose PB4 (it's that on my TFT but mine is TFT32), and in config.txt don't forget to put #enable auto off after print finish function (no:0; Yes:1）
>cfg_print_finish_close_Machine:1
I hope working for like for me.
Bonne chance (je peux te le faire en français si tu veux ca sera plus facile pour moi).

Torro666, 2017-09-27:
Merci beaucoup!
i'm gonna try it!

Torro666, 2017-09-27:
Hey,

i connected Ramps D11 - 5V - GND to
                   PWC    Sig  - NC - GND

then the other PWC plug to the TFT28 PB4

i changed to "power_supply = 2" in marlin.

Still the same problem. 
I think i have to do something in the pins.h but i can´t find something like "activate servo pin11" in pins.h

Torro666, 2017-09-27:
I found a "ps_on pin" under pins_ramps_pin.h

Changed it to 11.

Same Problem...

nemocaledonie, 2017-09-27:
Je te le fais en français, plus simple que Google translate.
- Sur le PWC, les fils qui vont au bouton, sont connectés comme suit : Bleu (O) - Vert (I)- Jaune ©.
- Sur le TFT tu connectes le cable du JST  S0-G-NC, le S0 sur PB4 et le G sur le GND (ou G) sur le TFT, le NC n'est pas connecté.
- Sur la carte RAMPS tu connectes S1-G-V, soit Signal-Masse-5V
- le config.txt du TFT tu mets 1 sur cfg_print_finish_close_Machine sinon jamais elle ne s'arrêtera.

ATTENTION LE PWC NE FONCTIONNE PAS AVEC LA NOUVELLE VERSION DE MARLIN (ou alors je n'ai pas vu comment faire), la version la plus récente qui fonctionne est 1.1.4

-Sur Marlin : - Dans configuration.h  --> POWER_SUPPLY 2
                       - pin.RAMPS.h     --> le numero de pin que tu veux et où il est connecté le fils Signal S1 (pin 11 ou 12)
On est bien d'accord que tu as déclaré ta carte en début de configuration.h (ligne MOTHERBOARD).
Lorsque tout est bien connecté, une petite LED verte s'allume sur le PWC et lorsque tu appuies sur le bouton, elle s'éteint brièvement.
J'espère que ca marchera pour toi, pour moi c'est le cas.
Dernier truc les connecteurs ne sont pas de super qualité pour les pins sur les cartes, vérifie !

Torro666, 2017-09-28:
Hi,

i wired everything up again.
the green LED is on. 
when i push the button, the LED is off, the printer starts.
when i release the button, the LED is on and the printer shuts down.

i tried PS ON Pin 11 and 12

it´s Marlin 1.1.0

any new idea?
tried it with 2 PWC boards...

makerbase-mks, 2017-10-16:
#enable UPS?(yes:1; no:0)
>cfg_have_ups_device:0

#enable power detecting module(mks 220det:1; mks pwc:0)
>cfg_insert_det_module:0

#enable auto off after print finish function (no:0; Yes:1）
>cfg_print_finish_close_Machine:1

makerbase-mks, 2017-10-16:
Marlin settings:
#if MOTHERBOARD == 33 || MOTHERBOARD == 34 || MOTHERBOARD == 35 || MOTHERBOARD == 77 || MOTHERBOARD == 67 || MOTHERBOARD == 68

  #define LARGE_FLASH true

  #if MOTHERBOARD == 77
    #define X_STEP_PIN         54
    #define X_DIR_PIN          55
    #define X_ENABLE_PIN       38
    #define X_MIN_PIN           3
    #define X_MAX_PIN           -1   //2 //Max endstops default to disabled "-1", set to commented value to enable.

    #define Y_STEP_PIN         60
    #define Y_DIR_PIN          61
    #define Y_ENABLE_PIN       56
    #define Y_MIN_PIN          14
    #define Y_MAX_PIN          -1   //15

    #define Z_STEP_PIN         46
    #define Z_DIR_PIN          48
    #define Z_ENABLE_PIN       63
    #define Z_MIN_PIN          18
    #define Z_MAX_PIN          -1

    #define Y2_STEP_PIN        36
    #define Y2_DIR_PIN         34
    #define Y2_ENABLE_PIN      30

    #define Z2_STEP_PIN        36
    #define Z2_DIR_PIN         34
    #define Z2_ENABLE_PIN      30

    #define E0_STEP_PIN        26
    #define E0_DIR_PIN         28
    #define E0_ENABLE_PIN      24

    #define E1_STEP_PIN        36
    #define E1_DIR_PIN         34
    #define E1_ENABLE_PIN      30

    #define SDPOWER            -1
    #define SDSS               25//53
    #define LED_PIN            13

    #define BEEPER             33

  #else

    #define X_STEP_PIN         54
    #define X_DIR_PIN          55
    #define X_ENABLE_PIN       38
    #define X_MIN_PIN          -1 // unused for deltabot, was 3
    #define X_MAX_PIN           2

    #define Y_STEP_PIN         60
    #define Y_DIR_PIN          61
    #define Y_ENABLE_PIN       56
    #define Y_MIN_PIN          -1 // unused for deltabot, was 14
    #define Y_MAX_PIN          15

    #define Z_STEP_PIN         46
    #define Z_DIR_PIN          48
    #define Z_ENABLE_PIN       62
    #define Z_MIN_PIN          18 // autolevel for deltabot
    #define Z_MAX_PIN          19

    #define Y2_STEP_PIN        36
    #define Y2_DIR_PIN         34
    #define Y2_ENABLE_PIN      30

    #define Z2_STEP_PIN        36
    #define Z2_DIR_PIN         34
    #define Z2_ENABLE_PIN      30

    #define E0_STEP_PIN        26
    #define E0_DIR_PIN         28
    #define E0_ENABLE_PIN      24

    #define E1_STEP_PIN        36
    #define E1_DIR_PIN         34
    #define E1_ENABLE_PIN      30

    #if MOTHERBOARD == 68
      #define E2_STEP_PIN        23
      #define E2_DIR_PIN         25
      #define E2_ENABLE_PIN      40

      #define E3_STEP_PIN        27
      #define E3_DIR_PIN         29
      #define E3_ENABLE_PIN      41

      #define E4_STEP_PIN        43
      #define E4_DIR_PIN         37
      #define E4_ENABLE_PIN      42
    #endif

    #define SDPOWER            -1
    #define SDSS               53
    #define LED_PIN            13
  #endif

  #if MOTHERBOARD == 33 || MOTHERBOARD == 35 || MOTHERBOARD == 67 || MOTHERBOARD == 68
    #define FAN_PIN            9 // (Sprinter config)
  #else
    #define FAN_PIN            4 // IO pin. Buffer needed
  #endif

  #if MOTHERBOARD == 77
    #define FAN_PIN            8
  #endif

  #if MOTHERBOARD == 35
    #define CONTROLLERFAN_PIN  10 //Pin used for the fan to cool controller
  #endif

  **#define PS_ON_PIN          11    //****

XneddyX, 2017-11-09:
Hello,
I have the same problem with MKS GEN L + MKS TFT32 + MKS PWC.
i push the button the printer is starting, i release and the printer is going off...

i have change the config.txt
> cfg_print_finish_close_Machine:1

and pin_RAMPS.h
>   #define PS_ON_PIN        11

configuration.h
> #define POWER_SUPPLY 

Marlin is on 1.1.5
TFT32 on 3.0

hope anybody can halp me...

lisergio, 2017-11-30:
hi!
today I received my mks pwc board but no connection scheme comes.
someone can tell me, where I have to connect all the cables,
Thank you

makerbase-mks, 2017-12-01:
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT2.8-3.2/MKS%20TFT28%2032%20V3.0%20datasheet.pdf

makerbase-mks, 2017-12-01:
This document is detailed

lisergio, 2017-12-01:
my pwc version is 1.0, this pdf is for other version.... the connection of the reset button is the same in the two versions?

lisergio, 2017-12-01:
I have already connected it.
I have set #define PS_ON_PIN 11
I have connected the PWC board to the connector of the servo 11 but if I stop pressing the button, the power supply goes off

makerbase-mks, 2017-12-02:
You can measure the servo 11, and if the motherboard is properly configured, the servo 11 will output 5v.

lisergio, 2017-12-02:
Hi again!!!
The pin works fine (the ramps needs a jumper for the output of servos to work)  , have 5 v when the ramps turn on.
When push the button, the printer turn on, but when the print end, do not turn off...

lisergio, 2017-12-02:
Finally Works!!!
I explain what I had to do
Arduino MEGA, Ramps 1.4 ,MKS TFT28 ( v3.0) and MKS PWC 1.0 using Marlin 1.1.4

connections:

The  push button and TFT28:

[image]

[image]

Ramps , servo PIN 6 and jumper ( to activate servo power )

[image]

Firmware:

TFT 28:
[image]

Marlin:
configuration.h

[image]

pins_RAMPS.h

[image]

To finish, you have to put the gcode M81 in the scripts at the end of the printing

[image]

Now when it finishes printing, the printer turns off, and to turn it on you have to press the button for a few seconds to restart the printer

emphasize, 2017-12-12:
Non-latching switch, i presume?

lisergio, 2017-12-12:
Of course, it is the push button that comes with the pwc

emphasize, 2017-12-12:
The latest offers on aliexpress are stand alone PWCs

gomines, 2018-02-07:
Can this module be used with de mks Sbase?

lisergio, 2018-02-10:
being the same manufacturer I imagine that if it worked

gomines, 2018-02-10:
I know you used Ramps in this case, but if anyone has an idea how to make it work on the Sbase would be of much help! I opened a new issue for this with the connections i did and the lines to add on the config.txt for Smoothieware:

https://github.com/makerbase-mks/MKS-TFT/issues/112

voodorush, 2018-03-17:
Same issue with MKS Gen_L. I use pin 6 too since 11 is ocupied by my BLtouch. Power button acts only if you hold it. This PVC 2.0 module but i believe the difference is only the led driver for 1.0 is 220v and the 2.0 uses 12v.
I can turn printer on/off with m80/m81. I also had to reverse 5v and gnd cables to servo 2 otherwise m80/m81 doesnt work too.

voodorush, 2018-03-17:
Ilsergio: you might want to add G4 S600 before M81 so the printer will wait 10 minutes to cool down the hotend before powering down.

emphasize, 2018-03-17:
> Ilsergio: you might want to add G4 S600 before M81 so the printer will wait 10 minutes to cool down the hotend before powering down.

Or M109 R80 (wait until cooldown temp of 80 degree is reached) followed by M104 S0, M81

Romano26, 2018-07-04:
Hello
since i have installed the mks pwc card on mks gen 1.4, i have breaks
  during printing with sdcard
can you help me please ? 
thk

handsomedude, 2018-07-04:
Try changing the baud rate of the LCD.  I had same issue.

I have ramps 1.4 Marlin 1.1.8. I had to set LCD to 115200 and set adruino to 250000 

That's the I lay way SD print does not stop anytime.

Hope it helps

Romano26, 2018-07-04:
thank you for your answer, but I have already tried; and my tft 2.8 screen does not communicate with my card mks gen 1.4 :-(

Romano26, 2018-07-05:
I found
it's the mks pwc card that creates the communication fault
I unplugged it :-(

nip2k2, 2018-07-15:
Hi,
Does anyone make MKS GEN 1.4 + MKS TFT32 + MKS PWC with marlin works?
Could you please help me?
My problem is that when I press the power button printer powers on, but when I release the button power goes off.
Plese let me know if you know how tol solve it.
I,ve try everything.
THanks in advance.

hartliebr, 2018-07-24:
Hello, config was MKS SBASE + MKS TFT32 + MKS PWC V2 and Smoothiware and the PWC worked fine.
Now with MKS GEN V1.4 + MKS TFT32 + MKS PWC V2 and Marlin V1.8.1, the problem is when I press the power button printer powers on, but when I release the button power goes off.
ANy clue?
Thanks in advance,
Bob

hartliebr, 2018-07-25:
YES !!!!!
IT WORKS !!!!
You’r the best, Many many thanks nip2k2 !

But you should comment this on GitHub, because you’r the only one TO KNOW…

I spend quite some time on it up to now and I was far from to find the solution.

Many thanks again and best regards,

Bob

De : nip2k2
Envoyé le :mardi, 24 juillet 2018 22:25
À : makerbase-mks/MKS-TFT
Cc : hartliebr; Comment
Objet :Re: [makerbase-mks/MKS-TFT] Tft28, pwc and ramps (#59)

Hi hartliebr, I had the same problem. Ive solved it swiching 5v and grond wire on gen 1.4. As I can guess the pwc conection diagram on datasheet is wrong (maybe because it for pwc v2.0). I looked into board diagram and realice that 5v and gronud were inverted to the colors of the conection diagram of pwc. Hope you understand what im trying to say.
I also change on marlin, configurarion.h POWER_SUPPLY 2 and on pins.RAMPS.h PS_ON_PIN 11.
Let me know if works for you. I spend a whole weekend with this hahaha.
Let me know if works.

coolt5, 2020-02-25:
A good news to everyone use MKS Gen L 1.4 with MKS TFT and MKS PWC v2
i mange to solve it .
it was the wiring from the bored MKS (PWC v2) to (MKS Gen L 1.4) the red(vvs) and black(Ground) was revers . so just swap them and it will work

makerbase-mks, 2020-05-27:
Hi! Our latest MKS PWC AliExpress page also has how to use it, you can refer to it, so I will close this question for the time being.

closed 2020-06-06 by makerbase-mks