makerbase-mks/MKS-TFT #60: MKS TFT 28 doesn't connect to RAMPS

martind-n, 2017-10-03:
Hi,

I'm facing an issue with mi MKS TFT 28. I'm trying to attach it to a RAMPS 1.4 board, however there is no communication at all. 
I tried to change baud rate on both boards but nothing. 
I'm also supplying 12V from an external PS (the same that powers RAMPS).
Funny thing is that running some USB connection (like pronterface) and being TFT28 connected, the screen shows the the temperature accordingly. I know I should use USB connected at the same time running TFT28, but it was just another test I made...

Is there an other thing to look at in order to get the board working?

makerbase-mks, 2017-10-16:
Is the serial line reverse?

martind-n, 2017-10-25:
No... On the MKS board there is only one position. On RAMPS I'm putin the red line pointing to power supply connector.

Ganriber, 2017-10-25:
I'm sorry, you probably wrote this letter to me by mistake and it is waiting for someone else. -- С уважением Макаров Павел Владимирович...

OskarLM, 2017-12-06:
I have the same problem. any solution?

martind-n, 2017-12-07:
Oskar,
I could solve my issue by changing the serial port on my Arduino (using serial2 instead of 0). That makes the screen communicates to printer, however there's another problem regarding printing from SD card.

dudisr, 2018-02-03:
i have the same problem, on ramps red cable same direction to power connector but i still cannot control by MKS

Devrim78, 2018-09-03:
Aynı  sorunu bende yaşıyorum.. Bir ürün sadece sattık ve oldu demekle olmaz maalesef, bir ürün satıldığı zaman arkasında durulması lazım. Ramps1.4 ile uyumlu yazılıyor fakat ne yaptıysak ürün çalışmıyor. Yazılan bütün her şeyi yaptım. Lütfen detaylı bir açıklamaya yaparsanız sevinirim....

Kodbee, 2020-05-19:
I have the same problem: MKS TFT28 V4 don't communicate with Ramps 1.4. any solution?

dudisr, 2020-05-21:
Even though is not a solution, i changed the ramps for skr1.3 the communication with board works flawlessly

makerbase-mks, 2020-05-27:
1. MKS TFT28 use DC12V power by DC terminal
2. Set the motherboard baud rate and connect AUX-1 correctly

Kodbee, 2020-05-27:
Thanks. I did, but still not connecting. This does not work in any way

martind-n, 2020-05-27:
Hi. What I actually did at that time was to change default por in Marlin for port #2 and then wired TFT TX to ramps pin #17 (RX) and TFT RX to ramps pin #16 (TX). By doing this I could get my tft and ramps working, however, the side effect is pronterface stop working, so mind your decision before doing this!

Kodbee, 2020-05-28:
Thanks Martind.n, but we need that it works fine with both ;) ... Does anyone solve this problem between MKS TFT28 V4.0 and Ramps?

makerbase-mks, 2020-06-06:
Should be caused by RAMPS problems. Need to see how the corresponding pins are distributed

Kodbee, 2020-06-06:
Thanks, I send you the pictures of how the the corresponding pins are distributed... It´s impossible to connect...
[image]
[image]
[image]

makerbase-mks, 2020-06-09:
MKS TFT28 AUX-1 pin
[image]

Kodbee, 2020-06-09:
The pins are correct. I have tried another cable, following the indications in the diagram that you have sent me, and it still does not connect.
[image]

Kodbee, 2020-07-29:
It´s impossible. MKS TFT28 4.0 **DOESN´T WORK** with Ramps 1.4. I am trying communicate both since 3 months ago, but it is impossible. MKS´s people don´t solve this issue. Money waste...

daniellezama, 2020-08-16:
Hi from argentina, I have the same problem Using Marlin 2.0 bugfix with a MKS SGENL V1.0 and a MKS TFT 3.5 with wifi. All in 250000 Bauds and no comunication. The display is on the wifi is conected, is in marlin mode, but no works the functions when i try to use the tft. With the reprap full graphics all works fine. Any have an idea???

Regards

Daniel

makerbase-mks, 2020-08-17:
@daniellezama
You can refer to MKS SGEN_L Wiki: https://github.com/makerbase-mks/SGEN_L/wiki/Display_MKS_TFT35

daniellezama, 2020-08-20:
> @daniellezama
> You can refer to MKS SGEN_L Wiki: https://github.com/makerbase-mks/SGEN_L/wiki/Display_MKS_TFT35

Yes I made this and works!! this is not commented in all tutorials that i have seen before. 

Regards

Printblaster