makerbase-mks/MKS-TFT #61: mks tft wifi problem

GhostPL2, 2017-10-14:
Hi I have problem because when I try connect through app MKSCloud for my 3d printer gets notification: "datas error" and "send data error" how fix it? And connect my 3d printer

makerbase-mks, 2017-10-16:
Are IP addresses and WIFI models configured?

GhostPL2, 2017-10-16:
Yes. I use You instruction for github. I use AP print mode

ValkyriaDelta, 2017-11-07:
I have the same problem, i think it may be the serial uart connection between the wifi module and the tft screen, because i can connect to the module but the screen won´t receive commands send by TCP/IP protocol (from repetier host or pronterface). Also the wifi settings on mks_config.txt don´t affect the wifi module, it´s IP by default is 192.168.4.1 and it won´t change even if i change it in the config file.

GhostPL2, 2017-11-07:
Producent send me new appCloud and it works

Hanrekakor, 2019-05-16:
> Yes. I use You instruction for github. I use AP print mode

where I could find this instruction?

GhostPL2, 2019-05-16:
Download link to pdf instruction:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.roboter-bausatz.de/media/pdf/04/6c/d6/MKS-TFT28-32-datasheet.pdf&ved=2ahUKEwi60ajq7KDiAhXpsIsKHZglDQEQFjACegQIBhAG&usg=AOvVaw3m2g1u2RiqvZr0JYeBZWnL&cshid=1558037200621