makerbase-mks/MKS-TFT #63: TFT32 input is reversed

thejordanclark, 2017-10-21:
On my MKS TFT32_L v1.0 with firmware V3.0.0  the input is mirrored.  If I click on the right side it registers the click on the left side.  All input is mirrored left <-> right.   Is there a way to reverse that?

makerbase-mks, 2017-11-09:
When and where did you download this firmware?

thejordanclark, 2017-11-10:
I downloaded it from GitHub,  This repo on 8/31.  So just over a month ago.

https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT2.8-3.2/Firmware/TFT28%2032_V3.0.0_English.zip

I have tried a few of the varients, currently it is loaded with "Red style" -> Retro

Math1010, 2017-11-30:
Same problem !
Have you find a solution ? 
Thanks

nargos, 2017-12-31:
Same problem with my v1.0 MKS TFT32 display.

:-(

thejordanclark, 2018-01-08:
@makerbase-mks Are you saying we can change or V1.0 TFT32 to a V3.0?  Do you have details about what the changes are?

thejordanclark, 2018-07-28:
I still have yet to find a solution.  I updated to the newest firmware released a couple month ago and it still has the same issue.  I would like to know how "you can change the hardware of TFT32_L V3.0" as stated by makerbase-mks.

closed 2020-06-06 by makerbase-mks

thejordanclark, 2020-06-06:
Is there no option for use that have MKS TFT32_L v1.0 on any current version of the firmware then?   You mentioned changing our "MKS TFT32_L v1.0" to "TFT32_L V3.0" but no response of how that is possible.

Any help would be appreciated.  Basically I currently have to touch the opposite section of the TFT32_L when ever I need to use it.

makerbase-mks, 2020-06-08:
Have you updated the firmware？The new firmware cannot directly support the old version of the hardware, mainly because the hardware does not support it, we are sorry!