makerbase-mks/MKS-TFT #64: TFT V3.0 Spanish language - Translation errors

tiwanacote, 2017-10-23:
Hi! Here I attach a summary of language translation errors. Many of them do not fit well and others directly make you to do not understand anything.
It's a shame, because in my country every time is used more mks board!
I hope they can correct the mistakes soon

Regards
Tiwanacote

[Corrections_language_V3.pdf](https://github.com/makerbase-mks/MKS-TFT/files/1408680/Corrections_language_V3.pdf)

makerbase-mks, 2017-11-09:
We've fixed it.

tiwanacote, 2017-11-16:
Dear friends, we have found some other errors. We attached PDF guide 
Thank you!
Best regards
Tiwanacote
[V2 errors.pdf](https://github.com/makerbase-mks/MKS-TFT/files/1480217/V2.errors.pdf)

Nitrosito, 2018-03-25:
I have a problem on the latest version... With language=5 (spanish)
I not see the text in the images (in the buttom).. i only see icons

My config:
```
>cfg_wifi_ap_name:Guarnited Red
>cfg_wifi_key_code:Sopadecucarachas2017.

#0:disable cloud service 1:enable cloud service
>cfg_cloud_enable:0
#cloud service adress
>cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#IP/mask/gateway
>cfg_ip_address:192.168.1.100
>cfg_ip_mask:255.255.255.0
>cfg_ip_gate:192.168.1.1

#--------------------------------------------------------------------------------

############## Function Customization #######################################

#user-defined function1 and 2.(display this button .  disable: 0; enable: 1)
>cfg_function_btn1_display:1

#command of user-defined function
>function_btn1_cmd:M84;

#set number of  "More" button 
>moreitem_pic_cnt:10

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G28 X0;

>moreitem_button2_cmd:G28 X0;

>moreitem_button3_cmd:G28 Y0;

>moreitem_button4_cmd:G28 Y0;

>moreitem_button5_cmd:G28 Z0;

>moreitem_button6_cmd:G28 Z0;

>moreitem_button7_cmd:G28;

#how many "More" button display on screen interface
>morefunc_cnt:3

#command setting on button1~6
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#---------------------------------------------------------------------------------
##### Display Customization(button and logo customized see User's Manual) #######

#the color of screen background
>cfg_background_color:0x000000

#the color of title text
>cfg_title_color:0xFFFFFF

#the color of temperature,fan,(etc) background
>cfg_state_background_color:0x000000

#the color of temprature,fan,(etc) text
>cfg_state_text_color:0xFFFFFF

#the color of file button
>cfg_filename_background_color:0x000000

#the color of file text
>cfg_filename_color:0xFFFFFF

#the color of background in printing 
>cfg_printingstat_word_background_color:0x000000

#the color of text in printing status
>cfg_printingstat_word_color:0xFFFFFF

#3D effects setting (disable: 0; enable: 1)
>cfg_BUTTON_3D_effects:0

#屏幕翻转180°(0:不翻转,1:翻转)
>cfg_OVERTURN_180:0
```

tiwanacote, 2018-03-26:
if you want text in buttons you have to edit images and include it

Nitrosito, 2018-03-26:
Really? Any pack have texts under icons? :(

Nitrosito, 2018-08-12:
@tiwanacote tiwanacote  Can you uploads your files with images please?

@tiwanacote puedes subir tus archivos con imagenes? No se como hacer un paquete con imagenes... El problema es que no se que archivo corresponde a que botón en que situación. Si subes el tuyo te lo agradezco mucho, veo que lo tienes muy bien montado

Un saludo

makerbase-mks, 2020-03-12:
Hi! Currently MKS TFT firmware is relatively stable, and we plan to open source the code for more enthusiasts.

closed 2020-03-12 by makerbase-mks