makerbase-mks/MKS-TFT #67: The problem with manual calibration of the hot bed.

Ganriber, 2017-10-26:
Good day! My printer endstop on the Z-axis is in the position of the Maximum. How can I make the table up to the nozzle when you select one of the calibration points?

[image]

[image]

padizar, 2017-10-29:
Это настраивается в прошивке контроллера, а не экрана. Что касается калибровки стола, то вот мои 4 точки для калибловки.

############  Leveling Function  ########################################

#leveling mode(manual:0; auto:1; conceal leveling button:2)
>cfg_leveling_mode:0

#the command of auto leveling (G29 is available for Marlin.While G32 is for  Repetier and Smoothieware)
>cfg_auto_leveling_cmd:G28;G29;

#the point number of manual leveling:(3,4,5 point available)
>cfg_point_number:4

#the coordinates of 5 point on manual leveling
>cfg_point1:20,20
>cfg_point2:280,20
>cfg_point3:20,280
>cfg_point4:280,180
>cfg_point5:180,180

Ganriber, 2017-10-30:
Добрый день! Тот текст что вы привели, как раз находится в config файле дисплея. Какой у вас принтер, типа моего? Стол поднимается к соплу, или у вас концевик Z стоит по Min?

padizar, 2017-10-30:
Принтер как у вас, когда стол внизу, это положение макс. Следовательно концевик настроен на макс.

Ganriber, 2017-11-08:
Padizar, как с вами можно связаться, пообщаться на тему ручной калибровки. Нужна помощь, сам разобраться не могу.

closed 2020-06-06 by makerbase-mks