makerbase-mks/MKS-TFT #70: TFT32 not recognizing my real time temps?

undertowx, 2017-11-04:
in Pronface everything works fine. When I use TFT it does not show what the temps are, just what the default settings should be ext: 0\180
                                          bed: 0\50

everything else seems to work, movement ,fans, file system,print?

anthonysleck, 2017-11-04:
Same issue here.

undertowx, 2017-11-04:
Any ides?

anthonysleck, 2017-11-04:
I am clean out of Ideas. I went back to my REPRAP GLCD that I know works.

Ditrow, 2017-11-07:
fw V3??? i have the same probleme too

undertowx, 2017-11-07:
No one seems to be able to help with this issue.

Ditrow, 2017-11-08:
I have solved the problem in mks_config to change the baud rate item to the one you set in marlin
I'm already printing

undertowx, 2017-11-08:
I am using Smoothie, the baud rates are already set the same, still doesn't show?

anthonysleck, 2017-11-08:
Ya the issue still exists on the TFT32 with the SBASE (AKA SMOOTHIE) even when the baud rate is synced. Right now it doesn't sound like it is going away. Returning product as I am extremely disappointed.

zen, 2018-01-13:
I have the same ridiculous problem