makerbase-mks/MKS-TFT #72: HLK client configuration

BlackWolf57, 2017-11-08:
I have a problem ton configure the HLK module with a TFT32 v3 and the latest firmware.
I want to configure it like a client, but the config file seems to be read partially.

`#############  WIFI Function  ##########################################

#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
>cfg_wifi_type:1

#wifi mode(AP:1; STA:2)
>cfg_wifi_mode:2

#wifi name and password
>cfg_wifi_ap_name:ap name
>cfg_wifi_key_code:wifi code

#0:disable cloud service 1:enable cloud service
>cfg_cloud_enable:1
#cloud service adress
>cfg_wifi_cloud_host:www.baizhongyun.cn
#cloud service port
>cfg_cloud_port:10086

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#IP/mask/gateway
>cfg_ip_address:192.168.1.5
>cfg_ip_mask:255.255.255.0
>cfg_ip_gate:192.168.1.1`

Same result with or without DHCP flag.
No cloud option visible on screen ???
Wifi status: Exception

The only way to make it run is to configure it in AP mode, and reconfigure the module by the web interface. But this is lost on reboot.

Tested on the box:
-SSID hidden or not
-MAC address filtering or not
-Changing encryption type

Tested on config file:
-cfg_wifi_type:0 => A cloud button is magically visible without the option active.
-cfg_wifi_mode:0 => Always STA Exception message.
-cfg_wifi_key_code Upper case or lower case it's the same.
-cfg_cloud_enable:0 => No change.
-cfg_ip_address:192.168.1.25 => Change the IP message with or without DHCP

Any ideas ?

makerbase-mks, 2017-11-25:
#wifi mode(AP:1; STA:0)
>cfg_wifi_mode:0

Sorry,our config file is no update,the wifi mode should be select 0.

BlackWolf57, 2017-11-25:
Thanks for your responce, but also tested and always the same "Exception" status.

Is there any limitation in the ap_name or the wifi_code ?
I have a ap_name with upper case, lower case and special characters, and my WPA2 key is a long key ( more than 20 digits ).

Also found a non # line in the config file, but make nothing !

revdaz, 2018-01-01:
I have the same problems, no solution?

BlackWolf57, 2018-01-12:
OK, made some test today.

I try multiple name and multiple size key on my phone, and the MKS-WiFi is connected like a client with an dynamic IP like 192.168.48.##

Now I want to connect it on my box, but my DHCP range is 192.168.1.10 to 192.168.1.150
192.168.1.1 is for the box.
Firmware v3.0.1 seems don't like this IP range.
Cannot change the IP of my box.

Manual configuration with the web interface of wifi module works well, so it's not the module or the security of my box.

closed 2020-06-06 by makerbase-mks