makerbase-mks/MKS-TFT #73: TFT32 MKS Gen-L crazy bed temp

CX-Piet, 2017-11-09:
Hi folks!
Any idea why the bed temp is jumping around? Sometimes over 10,000 ° C !!
The numbers flickering also (see below fan icon) don´t know if that is normal ..?
https://youtu.be/kYerpUElWV0

XneddyX, 2017-11-10:
Moin, ich würde mal die Sensoren tauschen um zu prüfen ob es an der Hardware liegt. Schwankt die Temperatur auch wenn du einen Computer als Host hast??? Ansonsten prüfe nochmal die Einstellung in Marlin welchen Sensor du eingestellt hast...
Hello, I would change the sensors to check whether it is due to the hardware. Does the temperature fluctuate even if you have a computer as a host? Otherwise, check again the setting in Marlin which sensor you have set ...

CX-Piet, 2017-11-10:
Moin! :))
Ich habe leider keine anderen Sensoren zum tauschen. Der Drucker lief aber vorher die ganze Zeit völlig unauffällig mit dem monochrom Display. Ich habe nur das Board+Display gegen ein MKS Gen-L mit TFT32 getauscht.

XneddyX, 2017-11-10:
Kannst du nicht den Sensor des hotend nehmen. In der Regel sind das die gleichen... Was für ein Drucker ist es denn???

CX-Piet, 2017-11-10:
Ist ein Creality Ender-4.
In der Anleitung bei Aliexpress stand ich soll bei Pins.h den Wert bei #ifndef TEMP_0_PIN
  #define TEMP_0_PIN -1 ändern und von -1 auf 11 stellen.
Damit lief der Drucker gut bis auf das Geflacker im Display halt.
Jetzt habe ich das mal wieder zurück gestellt auf -1 aber nun heizt das Hotend gar nicht mehr! Auch nicht wenn ich auf 11 zurückstelle. Die LED vom Mosfet leuchtet auf wenn ich den Befehl zu heizen gebe, aber die Temperatur ändert sich nicht und bleibt bei 23 Grad. Irgendwann geht dann die LED wieder aus, ich nehme an wegen time out....

XneddyX, 2017-11-10:
Hast du denn das Problem auch wenn der Host ein PC ist???

CX-Piet, 2017-11-10:
Wie meinst du das?
Den Befehl zum autotune gebe ich ja über den PC, funktioniert weder mit repetier host noch mit S3D. 
Und mir preheat über das Display geht's auch nicht.

closed 2020-06-06 by makerbase-mks