makerbase-mks/MKS-TFT #74: MKS TFT 32 Firmware will not update

usajet, 2017-11-10:
I load the firmware files onto the SD card as instructed and it only shows booting and not updating on the screen.

Ioshikazu, 2017-11-15:
Hi, the same here with me. Did you found any solution?

usajet, 2017-11-15:
Yes you need at least a 2 GB or lower SD card.

HugoW1975, 2017-11-29:
Hi, I just dowloaded these files:
https://github.com/robotsrulz/MKS-TFT
And put them on my 4GB USB stick, trying to update the firmware on my TFT because at the moment it is in Chinese. I start the system with the USB stick in place, screen connected to the MKS SBase V1.3. It just starts, no update. What should I do?
Cheers,
Hugo

usajet, 2017-11-29:
Use a 2 GB or lower SD Card.

zen, 2018-01-13:
Try SD card, not USB stick

MarkV3082, 2019-09-04:
Lifesaver thanks

closed 2020-03-12 by makerbase-mks