makerbase-mks/MKS-TFT #75: MKS TFT 32 Printing from SD card

martind-n, 2017-11-13:
Hi,

I'm trying to print from SD card slot on my TFT32 but the printing pauses for a while just after it starts or resumes. It does the same all the time, stops and re starts. Of course is impossible to achieve a printing with this behavior.
I already checked another thread about the same issue, related to GCODE M100. But after that, no new updates, that's why I'm starting a new topic.
Is there any plan to fix this on MKS side?
I'm running Marlin and TFT32 with FW 3.0.

Thirkell, 2018-03-11:
I too have this problem. I am running a MKS Sbase 1.3 with the latest version of smoothieware as well as the latest version of the firmware for the TFT from Makerbase. I have tried using the "lastest" firmware for the SBase from Makerbase as well but the same result. 
This flaw makes the product useless as one still has to print via some other means. Makerbase needs to fix this problem or open source the firmware and let the community fix this before they lose even more customers. I would return this if I could but it is more effort than its worth at this point. I have many local retailers in my country are no longer stocking this and advise people not to get it because of this issue.

martind-n, 2018-03-12:
Hi Thirkell,

So it doesn't work either with SBASE 1.3? I've just bought that board!
It's realy frustrating that same vendor can't get its screen working with its own control board.

I'll give it a try and let you know.

makerbase-mks, 2018-03-15:
@martind-n @Thirkell what's version of the tft32 and smoothieware.

Thirkell, 2018-03-15:
As mentioned, I am running V3.0.1 on the TFT Screen (https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.8-3.2/Firmware) and I am running the latest Smoothieware (https://github.com/Smoothieware/Smoothieware) I have tried different versions of firmware on both the SBase board and the TFT but it still gives the issue.