makerbase-mks/MKS-TFT #76: MKS TFT32 Filament Runout sensor does not beep

martind-n, 2017-11-14:
After triggering filament runout sensor, printing pauses but there's no alert. No beep, nothing.
As far as I could see on some youtube tutorials, it should start beeping until user fills up new filament.
Does anybody else have seen this behavior? 
I'm running Marlin 1.1.3 and TFT V3.0.

Thanks.
Martin.-

makerbase-mks, 2017-11-28:
We tested the version of TFTV3.0.0, and the buzzer would ring.

martind-n, 2017-12-01:
Mine doesn't... And the buzzer works fine becuase it beeps oon power on.
Any sugestion?

makerbase-mks, 2017-12-01:
where did you download the firmware?

makerbase-mks, 2017-12-01:
You can connect PB0 and GND, and if TFT32 is normal, the buzzer will ring

martind-n, 2017-12-07:
This is how I connected it but nothing happens... The printing stops but there's no sound alerting...

closed 2020-06-06 by makerbase-mks