makerbase-mks/MKS-TFT #79: 2 names texts for 1 icon on TFT32

terrible45, 2017-12-02:
Hi,
I have a MKS V1.5 with TFT 32 and after making a customs colors icons buttons, i have the text of 2 names that appear for 1 picture, why ? 

![screenshot_20171202-173309](https://user-images.githubusercontent.com/34192177/33517657-76e85516-d788-11e7-8018-75867bf6a09e.png

My picture have a text under, example with a Preheat:

 [image]

Who is possible to change this ? i didn't see  

Here my file config MKS_config.txt (style =>"SIMPLE")

##### Display Customization(button and logo customized see User's Manual) #######

#the color of screen background
>cfg_background_color:0x000000
#the color of title text
>cfg_title_color:0xFFFFFF
#the color of temperature,fan,(etc) background
>cfg_state_background_color:0x000000
#the color of temprature,fan,(etc) text
>cfg_state_text_color:0xFFFFFF
#the color of file button
>cfg_filename_background_color:0x000000
#the color of file text
>cfg_filename_color:0xFFFFFF
#the color of background in printing 
>cfg_printingstat_word_background_color:0x000000
#the color of text in printing status
>cfg_printingstat_word_color:0xFFFFFF
#3D effects setting (disable: 0; enable: 1)
>cfg_BUTTON_3D_effects:0
#屏幕翻转180°(0:不翻转,1:翻转)
>cfg_OVERTURN_180:0 

Thanks for your reponse 

)

makerbase-mks, 2017-12-04:
#multi-language(enable:1, disable:0)
>cfg_multiple_language:0

#languages setting
#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5). This configuration is valid when "cfg_multiple_language" is disabled.
>cfg_language_type:3

eight2stout, 2020-06-10:
hi,

I have an MKS Robin Mini and a TFT24
I updated the Firmware using the win8 example, and it shows text overtop of the icons, which have text on them,

I tried changing the config file as you specified. but its still showing text overtop of the icons.

#multi-language(enable:1, disable:0)

cfg_multiple_language:0

#languages setting
#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5). This configuration is valid when "cfg_multiple_language" is disabled.

cfg_language_type:3

as