makerbase-mks/MKS-TFT #80: Automatic pause for filament change

murdock62, 2017-12-02:
Hello,
Is there a solution for the printer to automatically pause for a filament change at a specific height when printing from the SD card.

Thanks for your help

makerbase-mks, 2017-12-04:
#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10
you can change the value of cfg_ZADD.

murdock62, 2017-12-21:
Nope that is not the problem. 
I want the screen to activate its pause function when the gcode on the SD card contains the M25 instruction. (Or another specific instruction ...)

makerbase-mks, 2017-12-30:
the screen will send the instruction when the gcode on the the SD.

captowel, 2018-03-25:
Murdock62 wants is this:
1) Gcode contains a pause   (M25 or M600)
2) MKS-TFT reads the M25 or M600 and AUTOMATICALLY pauses.  We do not want to start the pause touching the screen. 
3) The screen displays the pause and waits for the user to swap filaments 
4) The user pushes the resume button on the screen.

What we want, is an indefinite pause that can be issued by the gcode.  The MKS-TFT does not pause when using M25 or M600.  Both M25 and M600 are broken in your firmware.  G04 does work but a time interval must be entered.  We do not know how long the pause will need to be, therefore G04 is completely inadequate for this purpose.

Please let your user base know if an automatic pause command through the gcode is possible.   If it's not part of the standard gcode, please let us know if the screen supports additional terminal commands can be included within the gcode.

Thank you.

marcohaddad, 2018-10-26:
Hello,
I'm facing the same problem as Murdock62. 
Is the implementation of a gcode for pausing the MKS-TFT on a wish list, at least?
I'd love to have it. I guess many people would...
Best regards,

Wallaby07, 2019-07-21:
Has this been fixed, maybe the SD card direct to the MKS-TFT will ever work if it gets fixed in the firmware.
I want to use this plugin [https://www.thingiverse.com/thing:2077884](url) it can work with the M25, M0, M600 commands?

Rudy-Raw, 2019-08-23:
I am using marlin and is wondering if adding buttons with scripts will remedy this problem. I had to go and dig back up my 12864 display out of my junk box. Good thing I value junk and keep them around for a while.

closed 2020-06-06 by makerbase-mks

Flyingbroker, 2020-06-24:
Hello,
I bought an MKS Robin2 board for my own build 3D printer project, and I have the same problem, with pause at hight command, or filament chance pause. 
Is there a solution for this problem mean while??
I'm almost giving up, after I have done many tries, losing a lot of time, and thinking to switch to another board for my 3D printer project.
This is my last hope to keep working with MKS boards for my new 3d printer project.

Thanks for your help

nicolaunitn, 2020-09-30:
> Murdock62 wants is this:
>
> 1. Gcode contains a pause   (M25 or M600)
> 2. MKS-TFT reads the M25 or M600 and AUTOMATICALLY pauses.  We do not want to start the pause touching the screen.
> 3. The screen [...]

Hi, is there some update about this issue? I am looking for a solution for the same problem.
Thank you in advance!

JuggerRUS, 2020-12-11:
Why this was closed? I have MKS TFT24 and still have no options for automatic pause for filament change. Screen doesn't have any response to codes M25, M0 or M600. Built in filament change function works well, but it cannot be triggered via gcode.

Makerbase, we all need to have possibility to change filament at certain level of prining. For example, I want to print first 20 layers with one filament, then on 21st layer I want to change the filament and continue printing. How I supposed to do this via gcode? 

Now the only oprion which I have is to count layers and trigger filament change procedure manually via LCD. Not a best option if you ask me. Can you please add M600 command implementation into LCD?

manniefesto, 2021-08-24:
Hi, I am also having issues with the M0 command not working. Is there any movement on this?