makerbase-mks/MKS-TFT #81: mks tft32 v2.0 firmware update scrambled buttons

scooterharrell, 2017-12-05:
I updated my firmware to the latest and now all of my buttons look scrambled I've tried to revert back to V2 did not help

makerbase-mks, 2017-12-20:
anything pictures?

makerbase-mks, 2017-12-21:
you can update the firmware of v3.0.1

thejordanclark, 2018-01-17:
I don't know if it is related but I had a similar issue when I updated my firmware using a Mac to write the files to the SD card.  Something is messed up with how the Mac wrote the bin file to the SD card and how the MKS-TFT reads it which broke all the images.  I fixed mine by using a linux box in my case to write the files to the SD card.

makerbase-mks, 2018-01-31:
Where did you download the firmware and images.

makerbase-mks, 2018-01-31:
Where did you download the firmware and images?