makerbase-mks/MKS-TFT #83: Issues with MKS 32TFT connected to MKS Gen v1.4

d2xhale, 2017-12-06:
I'm having issues all the sudden with the MKS 32 TFT (LCD control panel) not sending code to the board. The printer currently will move using Pronterface linked using USB no problem. The MKS 32 TFT will power fine and respond to my touch browsing through menus but will not send code to the board. This problem just started and nothing has changed from when it last worked to now, Is there anyway to isolate the problem to help with problem solving?

zagranichnyy, 2017-12-10:
Repetier firmware?

makerbase-mks, 2017-12-20:
Is it normal to use it before?

d2xhale, 2017-12-21:
Using the latest Marlin firmware. Yes the panel was working perfect previously

wbarber69, 2018-02-18:
do you have the baud rate set properly?

closed 2018-04-26 by makerbase-mks

wbarber69, 2018-04-26:
Closed? Nothing is satisfactory here

Closed #83<https://github.com/makerbase-mks/MKS-TFT/issues/83>.