makerbase-mks/MKS-TFT #84: Missing customizable button with FW V3.0

atolala, 2017-12-07:
Hi

I Have try many things but impossible to add customizable button in Set menu, but also impossible in pause menu.

--This does not work :
#user-defined function1 and 2.(whether display this button .  disable: 0; enable: 1) 
>cfg_function_btn1_display:1

--This does not work too:
#how many "More" button display on screen interface
>morefunc_cnt:6
#command setting on button1~6
>morefunc1_cmd:G28;

Thanks

makerbase-mks, 2017-12-18:
the version of TFT32_V3.0.0 classic can not display button of function1. I test the morefunc_cnt is normal.the function2 is not supported in v3.0.0.

atolala, 2017-12-21:
Hi

Thanks for reply, ok i have found my problem with morefunc !

but can you make possible to display function1  classic ?

And can you check this https://github.com/makerbase-mks/MKS-TFT/issues/82

best regards

makerbase-mks, 2018-04-26:
classic type not display function1,we'll add it in next version.

closed 2018-04-26 by makerbase-mks