makerbase-mks/MKS-TFT #85: MKS TFT70 Firmware and Gcode preview

kimer2002, 2017-12-08:
1)Gcode preview does not work. 
how to fix it?

2)where to get the firmware?

makerbase-mks, 2017-12-20:
You need to use our software for slicing to get a preview.
this is the firmware and software:
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT70
https://github.com/makerbase-mks/Software/tree/mks-host

kimer2002, 2017-12-20:
Cool! Everything is working.
How is the picture encoded?
What are the sizes of the images (pixels) on the preview and during printing?
[image]

makerbase-mks, 2017-12-21:
the picture is bmp format.the sizes of the images is 100*100 and 200*200 (pixels) on the preview and during printing.

jdelolmosanchez, 2019-04-15:
Hello
I have a tft of 7 "and when I try to change the logo or any image of a button I see a screen with sloping lines and nothing is seen, I follow all the steps indicated in the manales but I can not solve it. I do not have any problem with the tft70 but this as you can haqcer megustaria perform it.
On the other hand I would like to ask if it is possible to include the image that the mks software gives in another slicing program like Cura.
Thanks so much, I wait for your answer.
Greetings.

jdelolmosanchez, 2019-04-17:
Hello, someone has the tft70 that can help me with the problem I have to change the logo and the buttons ????

Peepe57, 2019-09-08:
Link https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT70 is dead!

closed 2020-06-06 by makerbase-mks