makerbase-mks/MKS-TFT #88: repetier and TFT32 does not print

craftaz, 2017-12-28:
Repetier 0.92.8 and MKS TFT32 v3 does not start printing, only preheat, extrude and move xyz axes works correctly.

craftaz, 2017-12-28:
MKS TFT32 v3.0.1 same problem.

rafaljot, 2018-01-15:
Repetier 1.0, MKS TFT 28 v3.0.1 + MGS GEN 1.4, connectetd to AUX-1 @ 250000.
no communication at all 
```
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:2
#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1
#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:4
```

craftaz, 2018-01-17:
any news on this? or You'll just keep ignoring us?

craftaz, 2018-02-18:
So what? there's nothing about repetier firmware issue! this device is a fraud! Repetier 92.8 firmware is not compatible with this crap. If communication protocol issue (very old issue!) were easy fixable, they'd easily fixed it with recent update. Now it's a total crap, they're lazy developers and hopes all customers will use marlin firmware insded, wich is the only one that works :D 
p.s. please remove folowing lines from your code, after three months it insults me more and more:
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:2

dell9, 2018-02-18:
Hello. Try proposal from this issue:
https://github.com/makerbase-mks/MKS-TFT/issues/94#issuecomment-364855400

craftaz, 2018-02-18:
to comment WAITING_IDENTIFIER? where does anybody confirms it actually work? and why do i have to mess with my firmware at first place, when manufacturer tels me that it works with no firmware modification :)

dell9, 2018-02-19:
Is it too complex to try? My link contains confirmation, that this solution helped, but man faced with another problem related to delays in printing. You need to check yourself and place information here

makerbase-mks, 2018-02-27:
@craftaz @rafaljot 
If it's v_0.92.8,you need to config the firmware of the repetier as bellow:
// ################# Misc. settings ##################
#define ACK_WITH_LINENUMBER 1
#define WAITING_IDENTIFIER "wait"

If it's v_0.92.8,you need you delete the code in the gcode.cpp:

else if(lastLineNumber && Unable to render embedded object: File ((hasM() && M == 117)) { // once line number always line number) not found.
if(Printer::debugErrors())
{ Com::printErrorFLN(PSTR("Missing linenumber")); }
requestResend();
return;
}

tsmaa, 2018-03-04:
#No printing MKS TFT 32 v3.01 Marlin #119

I confirm the print problem for Marlin 1.1.8 + MKS TFT 3.2 v3.01

No solution at the moment

madias123, 2018-03-19:
Haha, I had to laugh!
Nearly I bought this display, but I checked github first.
Closed source? Are you kidding us? 
Nobody in the 3d-printer scene will give a fuck to closed source.
I feel sorry to all customers with problems, really.
I'll guess that the "author" used many open source components in the code.
So a message to all chinese developer (or copier): WE DON'T GIVE A FUCK TO YOUR CLOSED SOURCE DEVICES BASED ON OPEN SOURCE HARDWARE AND/OR SOFTWARE

closed 2018-04-26 by makerbase-mks