makerbase-mks/MKS-TFT #93: M-OFF button missing

marcohaddad, 2018-01-07:
I have just updated my MKS-TFT32 to firmware 3.0, and decided to stay with the classic style.
I wonder why there is a blank space on the Settings screen, where should be the Motors Off button, and there is not such function in any where else.

makerbase-mks, 2018-01-30:
the mks_tft32 firmware 3.0 display the M-off button only  on the simple and retro style.You can configure instruction of "M84" in the More button.

Jaka57, 2018-02-04:
No, I can't get to display for M-off button on retro style.
Firmware v.3.0.1, MKS TFT v.3.0

Jaka57, 2018-02-04:
bmp_function1.bin and bmp_function2.bin in blue style are the same but in red style they are as suppose to be.

makerbase-mks, 2018-02-05:
We've updated it.

marcohaddad, 2018-02-05:
The update was for the hole thing or just the reds button bmp?
Thanks!