makerbase-mks/MKS-TFT #94: MK4Duo + TFT32 - SD printing not working

dell9, 2018-01-07:
Hello!

It is not possible to use SD card print feature of TFT32 with the latest MK4Duo (v 4.3.3) firmware. After file selection from SD card and print start confirmation, TFT issues command N-1 M110*15, MK4Duo sends the response "ok\r\n" (I've removed #define ADVANCED_OK in the MK4Duo configurator to get simple responses with not details), then TFT just sends M105 getting response "ok T:23.1 /0 B:23.3 /0 @:0 B@:0\r\n" and no more actions done/commands sent by TFT32. 

Please advise response, that is expected by TFT from MK4Duo on N-1 M110*15 command? And please fix issue with SD card printing.

dell9, 2018-01-11:
If you do not have time to fix communication protocol - please share peace of code related with sd card printing and response processing. I can check and add required logic to support MK4Duo and Repeater firmware.

ozzysv, 2018-02-01:
same problem
developers are you here? fix the protocol of exchange

or do you want to say with your silence that your display works only on its own private protocol?

dell9, 2018-02-01:
Forget it. Developer not able to fix support of Repeiter firmware for a year. See bug https://github.com/makerbase-mks/MKS-TFT/issues/38 and https://github.com/makerbase-mks/MKS-TFT/issues/88. The latest Marlin is not supported too, see https://github.com/makerbase-mks/MKS-TFT/issues/56

And everywhere just ignoring from developer. Closed firmware, money for manufacturer, lost time and money for us.

makerbase-mks, 2018-02-02:
Why not used Marlin firmware ,we are not test the TFT32 for MarlinKimbra.

makerbase-mks, 2018-02-02:
Why not use Marlin firmware ,we are not test the TFT32 for MarlinKimbra.

dell9, 2018-02-02:
MarlinKimbra is based on Marlin and has the same communication protocol. Pure Marlin is not used since it does not support 32bit board. I am talking about stable version of Marlin. Furthermore, looking to https://github.com/makerbase-mks/MKS-TFT/issues/56 there is problem with Marlin too.

Please share communication logic/details. This is small part of code... Let us to adjust it and fix issue. You will get advantages only. We will get working product with our hardware.

makerbase-mks, 2018-02-03:
You need set the NO_TIMEOUTS  to 0:
#define NO_TIMEOUTS 0
I can print it as soon as I modify it.

dell9, 2018-02-03:
Thank you. This helped to me. Printer was able to print from SD card with the following settings of MK4Duo:
#define NO_TIMEOUTS 0
//#define ADVANCED_OK - disable

closed 2018-02-03 by dell9

dell9, 2018-02-12:
No, this is MK4Duo firmware

citkit, 2018-02-12:
hıımmm.
have you ever seen what ı should do on repetier firmware to start sd print tft32?
thank you for very fast reply :)

dell9, 2018-02-12:
Checked the firmware files. You can try 
//#define WAITING_IDENTIFIER "wait" - comment this line

citkit, 2018-02-12:
oowww ı cant not believe we have in progress. the print start :)  however it move a little  and wait , move and wait,  do you know what is the problem? the boudrate 115200

dell9, 2018-02-12:
No idea. May be problem with gcode. How it has been prepared? Share sample of used gcode

citkit, 2018-02-16:
ı easily print the same gcode with the same mks by using reprap lcd and ramps 1.4 and no problem.

dell9, 2018-02-16:
Do not know the reason. **makerbase-mks**, please advise if TFT32 has debugging mode to check problem with REPETİER firmware described by **citkit**?

**citkit**, anyhow, try to update REPETİER firmware to the latest version

makerbase-mks, 2018-02-27:
what's version of your Repetier firmware?

makerbase-mks, 2018-02-27:
@citkit what's version of your Repetier firmware?

citkit, 2018-02-27:
the latest version which is 0.92.9 version.

Thirkell, 2018-05-04:
I have the same issue where the print pauses intermittently. I am using an MKS Sbase board with the latest smoothie build and the latest firmware on the TFT from makerbase. It seems to be a very common problem, there are several issues on it on forums and on git. No proper response from Makerbase or apparent attempts to fir the problem. This business will fail if they dont open source the code soon for the community to fix