makerbase-mks/MKS-TFT #95: MKS TFT 3.2 and  HLK-RM04 WIFI Module

Spudsuds, 2018-01-07:
Can someone please explain the usefulness of the  HLK-RM04 WIFI Module? I cant get it to work with anything but MCSCload APP. Even then it only works from the LAN setting. I was hoping I would be able to send files directly to my printer instead of having to put them on a SD card first. Am I doing something wrong or is this all that is possible?

Thanks

thejordanclark, 2018-01-17:
the HLK-RM04 is basically a serial port TCP/IP gateway.  So you can connect to your printers serial port via WIFI instead of using a USB or serial cable as long as your printer control software supports a TCP/IP socket.  I.E I primarily use mine from the pronsole app (part of the printrun) it is handy for me to interact with the printer that way.

Spudsuds, 2018-01-17:
I cant get it to communicate with pronterface or repetier.  It connects to both but will not allow me to control the printer. I was hoping it would allow me to transfer files to the printer for printing but apparently not.