makerbase-mks/MKS-TFT #99: Translations/corrections

KuroiSabato, 2018-01-15:
I do want to translate the menus to italian and German. And also to correct some English errors...
how can I do it?

KuroiSabato, 2018-01-24:
no support?

MS1987, 2018-01-29:
Which lcd do you use? 
With TFT28/TFT32 you can ask makerbase to custom design for you.
With TFT24 , you can config through the config file.

KuroiSabato, 2018-01-29:
for the tft32.
ask for them? Don't think they can write on Italian and German... (not even English... still have "CANCLE" written on some screens...)
Anyway... seems that they are not concerned to answer my question. 

good products, but the technical support leaves to desire...

MS1987, 2018-01-30:
@KuroiSabato You can provide relative translation for them.

makerbase-mks, 2018-01-30:
You can provide relative translation for us and custom design the Italian of TFT32.

KuroiSabato, 2018-01-30:
I need, if possible, a list of all the commands to be translated. can you send me?

makerbase-mks, 2018-01-31:
Specific requirements and expenses you can contact:：shelly@makerbase.com.cn

makerbase-mks, 2018-01-31:
you can send the specific requrement to the email : shelly@makerbase.com.cn

KuroiSabato, 2018-01-31:
ok... will do. thank you

KuroiSabato, 2018-02-02:
it would be nice, if someone answer the email...

makerbase-mks, 2018-02-03:
give me your email.

makerbase-mks, 2018-02-03:
Give me your email to :engineer1@makerbase.com.cn

KuroiSabato, 2018-02-06:
email sent

KuroiSabato, 2018-02-09:
got the mail. will translate and send you asap.
thank you!

closed 2018-03-08 by makerbase-mks

FedericoBroglia, 2018-07-26:
Please May i have it too? I'm Italian I need translation for MKStft32 please help me

Laleyenda83, 2018-10-23:
Can i have the file too then i translate it to danish :)