makerbase-mks/MKS-TFT #100: MKS TFT32

medicshort87, 2018-01-19:
Any reason why after firmware update it is stuck on the booting screen?

seabassattitude, 2018-01-19:
I'm not sure if this is allowed or not, someone please let me know if it isn't. I'm including a link to a YouTube video that walked me through the firmware flashing and configuring process that fix this problem for me.

https://youtu.be/fmEB93af4o8

MS1987, 2018-01-29:
1. You have updated the wrong version of firmware.
2. Accident happenned when you were updating.

glucz, 2018-02-04:
I'm stuck on the booting screen too. I'm quite sure I updated the right version. How can I update again? it goes into the booting loop immediately when I power it on

seabassattitude, 2018-02-04:
This is same problem that happened to me. I posted Youtube link above. I followed this procedure to configure and flash firmware. Once you get the firmware flashed correctly, it will load normally. If it continues to only show BOOTING, then you haven't done it right.  Follow procedure in video. Once you have correctly updated firmware, this thing is plug and play. I really like this interface over the scroll knob printer came with.

glucz, 2018-02-04:
Thanks. I saw that. The problem is that I was stuck on the booting screen. It was not updating the firmware even when I rebooted with the SD card that had the update. I ended up using a different SD card as suggested by someone else.

seabassattitude, 2018-02-04:
I also switched SD cards. I hope you got it working now.

closed 2018-08-16 by medicshort87

dpatricio, 2020-06-11:
Switching cards didn't work for me. Has anybody found the bin to flash the bootloader? I guess that's the only option...