makerbase-mks/MKS-TFT #101: Power Outage to Continue not working

gomines, 2018-01-19:
Hello, when i power off my printer and power it on again, after hitting the reset button the extruder starts to preheat but when the bed begins to heat it shows 0° and the printing doesn't start, it just stays frozen.

Any ideas what the problem could be? Regarding firmware i've looked for any option that could make the bed go off but found nothing.

gomines, 2018-01-28:
Well i found out that with older versions (for ej: V2.0.1) the function "Power Off-Resume Printing" did work! Seems that there is something wrong in the latest version V3.0.1. 

Hope you can fix this issue and upload a new version!

MS1987, 2018-01-29:
@gomines Which hardware version of mks tft do you use?

gomines, 2018-01-29:
Hello, i have the MKS tft32 V3.0.0. I began using an SD card and now i’m using a USB.
Currently i changed the firmware to V2.0.1 and using the USB was able to use the “Power off-resume” feature.

makerbase-mks, 2018-01-31:
We tested the firmware of TFT_32 v3.0.1 and did not show any problems that you say. Could you please send us your motherboard firmware?

gomines, 2018-02-01:
I have the MKS Sbase and just tried both firmwares from Smoothieware's Github repository [https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin] with de TFT_32 v3.0.1 and experience the same problem. 
What firmware do you use?

makerbase-mks, 2018-02-02:
I use the Marlin,can you send your gcode file to me ,I want to test it with Smoothieware.

gomines, 2018-02-03:
The MKS Sbase can't be use with Marlin, only with Smoothieware. 

I used the firmware you have on your Github page and the "Power Off Resume" did work but things sush as motor direction and motor's speed changed using it (don't know why,perhaps it is an old version). I would like to use the latest versions of Smoothieware and still have this function of the TFT. Hope you can test this!
[Cauldron_Small.zip](https://github.com/makerbase-mks/MKS-TFT/files/1692311/Cauldron_Small.zip)

gomines, 2018-02-10:
Well with no feedback from my latest comment, i show my update: 

The function "power off-resume printing" DOES work with the firmware V3.0.1 on the TFT32 but ONLY if i set on the mks_config.txt "2" extruders instead of "1" as follows:

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:2

Now during a printing, i turn OFF the printer, turn it ON again, press Resume button, Extruder1 preheats, printer does G28XY, Bed preheats and finally printing resumes.

Don't ask me why it just worked when i changed it to 2 extruders.  #issueclosed

closed 2018-02-10 by gomines

vermak7033, 2019-11-30:
Hello Everyone,

I am facing the same issue, I am not able to resume the print in MKS TFT35. I have tried the latest firmware which is 1.0.2. I am using SD Card for printing purposes.

Also, I am not using any Power Outage detection module. I have tried the same thing in MKS TFT32 and MKS TFT70, and it is working perfectly fine in both the LCDs.

Please let me know what can be done.

makerbase-mks, 2020-04-01:
@vermak7033  Hi! MKS has release new firmware of mks tft35, you can update it.
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT3.5/TFT35_V1.0.5%20Release%20file.rar