makerbase-mks/MKS-TFT #105: Wifi Config is not updating

davigre, 2018-01-24:
I was using firmware 3.0.0 (English - Blue style - Retro ) with Wifi ( MKS HLK-WIFI V1.1 ) and was working ok.

I upgrade to 3.0.1 and noticed that I lost Wifi Connection ( as Client[sta] ). 

I start searching and noticed the following:

1. When upgrading files names changes:

     pic folder name: "mks_pic" -> "bak_pic"
     config file name: "mks_config.txt" -> "CONFIG.CUR"
     firmware file name: "mkstft28.bin" -> "MKSTFT28.CUR"
     Wifi Firmware file name: "MksWifi.bin" -> ( dont changes !!! ) 
     
2. The option in config file "cfg_wifi_mode:" don't changes. Always is in STA mode. I try both ( 0:sta and 1:AP ). "cfg_wifi_ap_name" & "cfg_ip_address" works ( In WIFI setting changes accordingly)

3. As STA with (of course) correct SSID and pass it don't connect to my router. This was working fine with 3.0.0 version.

4. I try to downgrade firmware ( to 3.0.0 ) but the same problems.

5. I reset the HLK-WIFI ( pressing the red button ) and it show his SSID. I enter to the WEB URL and manually change to my router SSID. It connects!!, so is Not the wifi module problem.

6. I try other variations ( WIN8 and English-Blue Style-Classic ) and the same problems occurs.

Please help.

Thank's.

[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/1658198/mks_config.txt)

makerbase-mks, 2018-01-27:
Is it normal to use the WiFi module in the 3.0.0 firmware?

MS1987, 2018-01-29:
@davigre As you are using "MKS HLK-WIFI", you should config those in the config file:
```
#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
>cfg_wifi_type:1
```
That should be work!

ajmarmar, 2018-09-20:
Hi, I've a MKS TFT 3.2 V4.0 with wifi module, and never connect with my router. the state STA is Disconnected.

MS1987, 2018-09-21:
@ajmarmar , please list the content about wifi in your "mks_config.txt" . And have a look the wifi firmware version through the TFT ("Set"——>“Wifi”)