makerbase-mks/MKS-TFT #106: TFT 32 beeps, no picture

Lino77, 2018-01-31:
TFT 32 beeps, no picture whats that

makerbase-mks, 2018-02-01:
Have you updated the firmware?

Lino77, 2018-02-01:
Yes to the lautest Last week an fit worked but yesterday if i switched the Printer on they Beeps 
On the MKS sbase ans also on the mks Rumba+

dudisr, 2018-02-03:
on 2.8 im having the same issue. I couldn't completed fix but instead of unzip then copy to the SD card i unzip directly in SD. This way mostly of the pictures appear, but still there are some shadows in screen or for ex. back button that never show its icon. Is usable but not fixed. Still waiting the guy to come up with a new version that show all  pictures.

Lino77, 2018-02-03:
dont work only 
beep .... beep,beep...beep,beep...beep,beep...beep,beep...beep,beep...beep,beep...

Lino77, 2018-02-10:
what is now with my tft

mks83, 2018-06-01:
Hi, I have a MKSTFT28 v3 that does exactly the same thing. Can someone please post the fix or verify that the board is defective. I have found a few threads like this but no one ever posts an outcome.

karandex, 2018-06-03:
same problem here. it was working on the older firmware. it first showed some glitches on display. so decided to update it to old firmware and this started to happen. At least provide us with a compiled base bootloader so we can burn it via st link. you guys are very stubborn, you already have it closed source. at least help us fix out on our own.

Popeyef5, 2019-02-07:
Same problem. Did anyone find a solution?

ascarons22, 2019-02-12:
Yo brickee la mia a proposito para reparar este problema en otra tft,lo solucione con un stlink y reconfigurando el stm32,mi tft volvio a funcionar correctamente.

dpatricio, 2020-06-11:
> Yo brickee la mia a proposito para reparar este problema en otra tft,lo solucione con un stlink y reconfigurando el stm32,mi tft volvio a funcionar correctamente.

tienes algún tutorial sobre esto?

G2Barbour, 2020-07-08:
Being that you guys have open-sourced the firmware now, is there any way you would be willing give some answers about this issue?